bb-qq/aqc111 #86: DS920+ DSM 7.1.1-42962 U1 -aqc111 (SABRENT NT-SS5G)-geminilake-1.3.3.0-9.spk -> Fix: Edit the Network Interface to reset the IP #84

KipiKland, 2022-11-08:
I initially connected the USB Ethernet adapter on the back port of the DS920+, and all was working fine.
I changed then to front port and even after a NAS restart IP advertised was the same but could not ping it. I confirm i had to apply the same fix in a different post, that was to change IP to another and revert back to the initial one for the IP to work.

arthurlockman, 2022-11-17:
I'm on the same platform (same DSM version too) with the QNAP adapter. I have a recurring issue where the interface just seems to disable itself until I login and disable/enable the driver. No connectivity though the ports on my switch appear to be lighting up as if it's connected. Unplugging the cable doesn't make a difference - I have to reset the driver.

arthurlockman, 2022-11-17:
Of course! Here's the last 24 hours or so. The failure was just before 8:30PM pacific yesterday. https://gist.github.com/arthurlockman/cf99a99cfc9d37225370d2acfd06024a

bb-qq, 2022-11-19:
Thanks, @arthurlockman. In your case, changing the USB port to which the adapter is connected may improve the symptoms.

@KipiKland, in your case, there seems to be a problem with the driver's handling of restoring IP settings, so logs are needed to investigate.

arthurlockman, 2022-11-19:
Thanks @bb-qq, I'll give it a shot. I had it plugged into the front USB port but I've moved it to the back. I'll report back if that worked. If it's going to fail again it should fail in the next couple days.

arthurlockman, 2022-11-19:
Hmm, ok, so moving it to the back is a non-starter. For whatever reason when it's plugged into the rear USB port, Bonjour (avahi) refuses to work correctly. It never registers a name, it just keeps counting up trying to register. It only works correctly when the adapter is plugged into the front.

arthurlockman, 2022-11-29:
@bb-qq i managed to fix bonjour on the rear port, so I switched the adapter over there. It worked totally fine for several days but this morning exhibited the same symptoms again. I took a snapshot of the logs: https://gist.github.com/arthurlockman/7ff258245406503e6c64b1347aa09828 The failure was around 07:30AM pacific. Any ideas about what's going on?

bb-qq, 2022-12-03:
Can you please paste the output of `dmesg` as I need the kernel logs to diagnose the symptoms?
Also [Troubleshooting|https://github.com/bb-qq/r8152/wiki/Troubleshooting] page might help to improve the situation.

arthurlockman, 2022-12-03:
Sure, if/when it happens again I'll grab a snapshot of dmesg. It seems to be working fine right now, but in dmesg I do see a lot of these errors going on: 

[image]

Maybe these have something to do with it?

bb-qq, 2022-12-04:
The logs from aqc111 `kevent 2 may have been dropped` can be ignored if they are infrequent, but from the `usbfs` one it looks like something bad is happening with this UPS device.

Could you try unplugging all USB devices except the Ethernet dongle to isolate the problem?

arthurlockman, 2022-12-04:
Yeah, I had the same thought and unplugged it this morning. Will report back when it happens again.

bb-qq, 2022-12-10:
Hmmm///
Changing the USB port to connect the NIC with the other USB devices disconnected does not change the result, right?

bb-qq, 2022-12-31:
I would like to hear any updates on the situation.

arthurlockman, 2022-12-31:
No updates right now unfortunately. I can't get another NIC for quite a while. Deliveries are a month out, no one has one in stock. I'll update if I manage to get another one.

arthurlockman, 2023-01-01:
I'm trying out one thing in the mean time. I had a hunch that it might be power related - perhaps the dongle is drawing too much power sometimes and shutting the USB port down. I've reinstalled it with a much shorter ethernet cable to the transceiver on the other end. We'll see if that makes a difference. If it does a more permanent solution might be to use a powered USB hub to give the dongle more juice.

closed 2024-06-15 by bb-qq