bb-qq/aqc111 #87: DS220+ SPK can't work

bg2vcj, 2022-11-16:
DS220+ 
SPK can't work
DM:7.1.1-42962
QNA-UC5G1T
use aqc111-geminilake-1.3.3.0-9.spk
After installation, does not work.
“driver for aquantia AQC111U based usb ethernet adapters [dsm7 note] the installation will fail the first time”
please help me

thanks

bg2vcj, 2022-11-17:
remark：
plug out QNA-UC5G1T  USB
aqc111 SPK  can runing
(Can be stopped and run)

plug in QNA-UC5G1T USB
aqc111 SPK  can't run
Startup failed
Needs to be fixed

bg2vcj, 2022-11-24:
[aqc111-geminilake-1.3.3.0-10.spk](https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-10/aqc111-geminilake-1.3.3.0-10.spk)  working
thanks

bg2vcj, 2022-11-29:
new question:
NAS write rate: 250-260M/s
Reading rate: 150-160M/s
please fix

bb-qq, 2022-12-03:
I close this issue since the issue was resolved.
I would appreciate it if you report the test result in this thread.
https://github.com/bb-qq/aqc111/issues/79

closed 2022-12-03 by bb-qq