bb-qq/aqc111 #88: DS218+ with Sabrent NT-SS5G - install, shutdown the NAS and the interface dissapears

dvelayos, 2022-11-16:
Model (platform): DS218+ (apollolake)
DSM version: 7.1.1-42962 Update 1
Driver version: 1.3.3.0-9
Ethernet device: Sabrent NT-SS5G

Working status: not working

i've followed the installation steps, entered on the network DSM configuration and i've seen a LAN 2 available (so at first, the networks seems to be created) but, instead of configuring the new network, i've shut down the NAS (by mistake :-(), and when the NAS has started, the LAN 2 network had disappeared. i've tried reinstalling the driver, with no luck

The logs (by the way, there is a typo on the 1.3.3.0-9 release: its /var/packages/aqc111/var/log/ , not /var/package/aqc111/var/log/ ) seems to try to use a eth1 / ovh_eth1 interface, but it's not shown on ifconfig / ip address

if i try to start the package, fails with this message:

`synopkg start aqc111
{"action":"start","beta":false,"error":{"code":272,"description":"Failed to run script, script=[start]"},"finished":true,"language":"enu","last_stage":"prepare_start","package":"aqc111","pid":19608,"scripts":[{"code":1,"message":"","type":"start"}],"stage":"start_failed","status":"stop","status_description":"translate from systemd status","success":false,"username":"","version":"1.3.3.0-9"}

dvelayos, 2022-11-17:
(installing the 1.3.3.0-8 works, so i supose the problem is related with the fixes on 1.3.3.0-9)

bb-qq, 2022-11-17:
Could you paste the log here?

dvelayos, 2022-11-17:
yes, sorry.
[start-stop-status.log](https://github.com/bb-qq/aqc111/files/10034795/start-stop-status.log)

bb-qq, 2022-12-04:
Thanks.
Could you try the latest version?
https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-10

closed 2024-06-15 by bb-qq