bb-qq/aqc111 #89: Can't install 1.3.3.0-10 DSM-7.x on DS918+

sparkplug37, 2022-11-23:
### Description of the problem

I have the previous version of the driver installed on my DS918+ using the package and tried to install this new version 'aqc111-apollolake-1.3.3.0-10.spk', but I get the error:

Unable to install AOC111 driver because it runs with root privileges. Packages with root
privileges might execute commands as the root user and compromise your system
security. Please contact the package developer to change the privilege settings of the
package.

### Description of your products

Synology DS918+
Linux DS918 4.4.180+ #42962 SMP Wed Sep 21 10:56:47 CST 2022 x86_64 GNU/Linux synology_apollolake_918+
DSM 7.1.1-42962 Update 1

### Description of your environment

Connected to the network via a Netgear 10g Switch
Aquantia 5G USB Ethernet Adapter - Aquantia
Netgear XS508M
Cat 6 cable

### Output of ``dmesg`` command

Sorry too long.

### Output of ``lsusb`` command

|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F008A2CABE41617)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         2eca:c101:0101 00  3.20 5000MBit/s 896mA 1IF  (Aquantia Aquantia 5G USB Ethernet Adapter 000000E1)

### Output of ``ifconfig -a`` command

docker0   Link encap:Ethernet  HWaddr 02:42:5A:F5:1D:A7  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:5aff:fef5:1da7/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11170 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8617 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11531907 (10.9 MiB)  TX bytes:849733 (829.8 KiB)

docker-7e Link encap:Ethernet  HWaddr 02:42:04:8A:64:B3  
          inet addr:172.21.0.1  Bcast:172.21.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

docker2df Link encap:Ethernet  HWaddr 5E:19:FB:16:A4:90  
          inet6 addr: fe80::5c19:fbff:fe16:a490/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:21 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:2244 (2.1 KiB)

docker4a3 Link encap:Ethernet  HWaddr A6:AC:2C:F2:C4:88  
          inet6 addr: fe80::a4ac:2cff:fef2:c488/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:19 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:2064 (2.0 KiB)

docker720 Link encap:Ethernet  HWaddr 6A:18:80:0E:AD:97  
          inet6 addr: fe80::6818:80ff:fe0e:ad97/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:2502 (2.4 KiB)

docker9ab Link encap:Ethernet  HWaddr 9E:A7:8D:BE:F8:35  
          inet6 addr: fe80::9ca7:8dff:febe:f835/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:11170 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8617 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:11688287 (11.1 MiB)  TX bytes:849733 (829.8 KiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AD  
          inet addr:172.16.1.25  Bcast:172.16.7.255  Mask:255.255.248.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:194989 errors:0 dropped:0 overruns:0 frame:0
          TX packets:134396 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:193114161 (184.1 MiB)  TX bytes:28665895 (27.3 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:9A:D4:AE  
          inet addr:169.254.18.190  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 00:24:27:88:36:C1  
          inet addr:172.16.1.22  Bcast:172.16.7.255  Mask:255.255.248.0
          inet6 addr: fe80::224:27ff:fe88:36c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:6959 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2411 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1277568 (1.2 MiB)  TX bytes:1229624 (1.1 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:8324 errors:0 dropped:0 overruns:0 frame:0
          TX packets:8324 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:891545 (870.6 KiB)  TX bytes:891545 (870.6 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-11-23:
Sorry, I uploaded the wrong packages, so I re-uploaded the correct ones. Please re-download the package.

sparkplug37, 2022-11-23:
Thank you, that worked.

bb-qq, 2022-11-23:
Thank you for the quick response. I close this issue.

closed 2022-11-23 by bb-qq