bb-qq/aqc111 #92: Poor Transfer Speeds with SABRENT NT-SS5G and DS1019+

BlackReaper66613, 2022-12-20:
### Description of the problem

> I am able to reading at about 350 MBs but my write speed is about 90MBs

### Description of your products
> - DS1019+
>     - Linux SYNDATA 4.4.180+ #42962
>     - DSM 7.1.1-42962 Update 1
> - SABRENT NT-SS5G
### Description of your environment
> - Connected to 10G Switch
>     - CRS305-1G-4S+IN
> - cable category CAT6A
### Output of ``dmesg`` command
redacted too long
### Output of ``lsusb`` command
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0092A994CA6054)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-2         2eca:c101:0101 00  3.20 5000MBit/s 896mA 1IF  (Aquantia Sabrent 5G Ethernet Adapter 00000019))
### Output of ``ifconfig -a`` command
bond0     Link encap:Ethernet  HWaddr 00:11:32:9B:25:B5
          inet addr:192.168.40.40  Bcast:192.168.40.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe9b:25b5/64 Scope:Link
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:9000  Metric:1
          RX packets:3339 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5259 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:358367 (349.9 KiB)  TX bytes:3436818 (3.2 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:9B:25:B5
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:300085 errors:0 dropped:0 overruns:0 frame:0
          TX packets:61657 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:445684691 (425.0 MiB)  TX bytes:6942665 (6.6 MiB)
          Interrupt:125

eth1      Link encap:Ethernet  HWaddr 00:11:32:9B:25:B5
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:3424 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9433 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:412997 (403.3 KiB)  TX bytes:6725719 (6.4 MiB)
          Interrupt:126 base 0x6000

eth2      Link encap:Ethernet  HWaddr 00:24:27:88:45:03
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:11896 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2240 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:1129022 (1.0 MiB)  TX bytes:367368 (358.7 KiB)

eth2.40   Link encap:Ethernet  HWaddr 00:24:27:88:45:03
          inet addr:192.168.40.45  Bcast:192.168.40.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:2632 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2204 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:588104 (574.3 KiB)  TX bytes:334401 (326.5 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:158 errors:0 dropped:0 overruns:0 frame:0
          TX packets:158 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:23268 (22.7 KiB)  TX bytes:23268 (22.7 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

bb-qq, 2022-12-31:
Did you check the [Troubleshooting](https://github.com/bb-qq/r8152/wiki/Troubleshooting) page?

sparkplug37, 2023-02-19:
I, too, have noticed slow speeds using this interface. To test things I have run iperf3, which shows the expected speeds, but using an smb file copy its very slow at 10MB/sec. So in my experiments, I have discovered disabling the Synology smb service and re-enabling it gives back the smb speeds I had expected of approx 150MB/sec

bb-qq, 2023-03-18:
I don't know if this is related to your problem, but if the MTU values are not set properly at the source and destination, the performance may not be sufficient. You may also want to try setting the SMB protocol version to 3.

closed 2024-06-15 by bb-qq