bb-qq/aqc111 #93: DS923+ AMD Ryzen R1600 CPU DSM 7.x driver Supported

inflection-zz, 2022-12-24:
Recently, I upgraded to Synology DS923+.
But I can not find AQC111 DSM driver for AMD Ryzen R1600, even DS923+ with 10G Ethernet option.
Can you develope DS923+ AMD Ryzen R1600 DSM 7.x driver?
Thanks for reading.

bb-qq, 2022-12-31:
Please try this package.
https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-10/aqc111-r1000-1.3.3.0-10.spk

inflection-zz, 2023-01-09:
> Please try this package. https://github.com/bb-qq/aqc111/releases/download/1.3.3.0-10/aqc111-r1000-1.3.3.0-10.spk
Thanks for your reply I did not find r1000 package.  
But insufficient system partition space message is  displayed.
[image]
[image]

bb-qq, 2023-01-28:
Thank you for the report.
I close this issue since the driver is working.

closed 2023-01-28 by bb-qq

inflection-zz, 2023-01-30:
Thanks for your reply and r1000.spk.
It is working very well on synology ds923+.
Thanks, Best Regards