bb-qq/aqc111 #95: DS918+ apollolake 2.6.13-3 & Plugable USBC-E2500, missing from UI and eth2 not populated

sgreadly, 2023-01-04:
### Description of the problem

I followed the instructions on installing r8152-apollolake-2.16.3-3.spk and started the package successfully.

The problem is, eth2 is not showing up in ifconfig output (if i just run **ifconfig** without -a), and it's not displayed anywhere in the UI for me to configure. 

I'm not sure what exactly I did wrong, or didn't fully do, to get it configured correctly. 

### Description of your products

DS918+ - DSM 7.1.1-42962 Update 3

````
# uname -a
Linux S1 4.4.180+ #42962 SMP Tue Oct 18 15:05:01 CST 2022 x86_64 GNU/Linux synology_apollolake_918+
````

[Plugable USBC-E2500](https://amzn.to/3rIVuZw) - connected to the front port. The lights on the adapter are on indicating it's up and running.

### Description of your environment

Synology connected to my Ubiquiti switch, which my MacBook is also connected to.

Cat6 cable.

### Output of ``dmesg`` command

````
[ 2217.203925] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[ 2217.347055] usb 2-1: reset SuperSpeed USB device number 2 using xhci_hcd
[ 2217.418038] netif_napi_add() called with weight 256 on device eth%d
[ 2217.425302] r8152 2-1:1.0 eth2: v2.16.3 (2022/07/06)
[ 2217.430865] r8152 2-1:1.0 eth2: This product is covered by one or more of the following patents:
               		US6,570,884, US6,115,776, and US6,327,625.

[ 2217.448257] r8152 2-1:1.0 eth2: chip rev 14
[ 2217.452943] r8152 2-1:1.0 eth2: rx_buf_sz will be overridden to 16384
````

### Output of ``lsusb`` command

````
# lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-1         051d:0002:0106 00  1.10  1.5MBit/s 24mA 1IF  (American Power Conversion Back-UPS XS 1400U  FW:926.T2 .I USB FW:T2  4B1951P15092  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 650082A893DA5A45)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)
````

### Output of ``ifconfig -a`` command

````
root@S1:~# ifconfig -a
bond0     Link encap:Ethernet  HWaddr 00:11:32:8B:3E:89
          inet addr:192.168.xx.xx  Bcast:192.168.xx.255  Mask:255.255.255.0
          inet6 addr: fe80::xx/64 Scope:Link
          inet6 addr: 2407:xx/64 Scope:Global
          inet6 addr: 2407:xx/128 Scope:Global
          UP BROADCAST RUNNING MASTER MULTICAST  MTU:1500  Metric:1
          RX packets:53338 errors:0 dropped:0 overruns:0 frame:0
          TX packets:59101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12167171 (11.6 MiB)  TX bytes:30895928 (29.4 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:8B:3E:89
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:53338 errors:0 dropped:0 overruns:0 frame:0
          TX packets:59101 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:12167171 (11.6 MiB)  TX bytes:30895928 (29.4 MiB)

eth1      Link encap:Ethernet  HWaddr 00:11:32:8B:3E:8A
          UP BROADCAST SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth2      Link encap:Ethernet  HWaddr 8C:AE:4C:DD:41:D5
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:24618 errors:0 dropped:0 overruns:0 frame:0
          TX packets:24618 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:6493189 (6.1 MiB)  TX bytes:6493189 (6.1 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

sgreadly, 2023-01-04:
Ok so I had to restart the NAS twice to get it to work - i.e., to show the interface in the UI. 

After the first reboot, I noticed the "RTL8152" package was installed but not running. I started it (run) manually. 

I had the adapter connected to the back USB port, but moved it to the front USB port after reading your documentation, and that's where I ended up at with the above details.

I just gave it another reboot and yes now it's working as expected. 

Hope this helps, maybe to add somewhere in the instructions or docs as a "take care!" step.

bb-qq, 2023-03-19:
I added a note about a reboot may be necessary.

If you are still using the acc111 driver, I would be happy to have a report of how it works on this thread.
https://github.com/bb-qq/aqc111/issues/79

closed 2023-03-19 by bb-qq

sgreadly, 2023-03-19:
@bb-qq , do you mean aqc111? No worries I'll reply to the other thread tomorrow. Seems stable since! You can close this one off if you need to. Ta.