bb-qq/aqc111 #98: DS418j AQC111 driver does not start

powerpc970, 2023-01-15:
### Description of the problem

I installed the AQC111 driver as instructions, but the driver does not start.
Displayed "AQC111 driver failed to start. Please try again later or contact the package developer" in Package Center.
Pressing the repair button does not start and the same display repeats.

The r8152 driver is already installed and working properly.
eth1 is r8152 driver.

### Description of your products

> - the product name of your NAS model

NAS model: DS418j
DSM version: DSM 7.1.1-42962 Update 3
$ uname -a
Linux xxxxx 4.4.180+ #42962 SMP Tue Oct 18 15:07:11 CST 2022 aarch64 GNU/Linux synology_rtd1296_ds418j
Ethernet dongle: QNA-UC5G1T

### Description of your environment

PC: MacBookPro MacOS 12.6.2

### Output of ``dmesg`` command

[ 3643.230998] ata1: wake up from deepsleep, reset link now
[ 3643.555138] ata1: device plugged sstatus 0x1
[ 3652.028596] ata1: link is slow to respond, please be patient (ready=0)
[ 3653.908576] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[ 3653.920008] ata1.00: configured for UDMA/133
[ 3653.924389] ata1: wake up successful, the reset fail can be ignored
[11506.323694] ata1: wake up from deepsleep, reset link now
[11506.644447] ata1: device plugged sstatus 0x1
[11515.121392] ata1: link is slow to respond, please be patient (ready=0)
[11517.061370] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[11517.072542] ata1.00: configured for UDMA/133
[11517.076923] ata1: wake up successful, the reset fail can be ignored
[13034.509984] ata1: wake up from deepsleep, reset link now
[13034.829097] ata1: device plugged sstatus 0x1
[13043.304675] ata1: link is slow to respond, please be patient (ready=0)
[13045.124689] ata1: SATA link up 3.0 Gbps (SStatus 123 SControl 300)
[13045.132473] ata1.00: configured for UDMA/133
[13045.136897] ata1: wake up successful, the reset fail can be ignored
[15325.590176] audit: type=1325 audit(1673777865.704:33): table=filter family=2 entries=0
[15325.784191] audit: type=1325 audit(1673777865.904:34): table=filter family=2 entries=4
[15325.807342] audit: type=1325 audit(1673777865.924:35): table=filter family=2 entries=8
[15349.120015] audit: type=1325 audit(1673777889.234:36): table=filter family=2 entries=9
[15349.147640] audit: type=1325 audit(1673777889.264:37): table=filter family=2 entries=8
[15349.204041] Module [xt_recent] is removed. 
[15349.277784] Module [xt_iprange] is removed. 
[15349.316135] Module [xt_limit] is removed. 
[15349.375983] Module [xt_state] is removed. 
[15349.434905] Module [xt_tcpudp] is removed. 
[15349.478373] Module [xt_multiport] is removed. 
[15349.555791] Module [xt_LOG] is removed. 
[15349.631291] Module [iptable_filter] is removed. 
[15352.678792] audit: type=1325 audit(1673777892.794:38): table=filter family=2 entries=0
[15352.921524] audit: type=1325 audit(1673777893.044:39): table=filter family=2 entries=4
[15352.945144] audit: type=1325 audit(1673777893.064:40): table=filter family=2 entries=8
[15366.527668] mii: exports duplicate symbol generic_mii_ioctl (owned by kernel)
[15371.603783] audit: type=1325 audit(1673777911.725:41): table=filter family=2 entries=9
[15371.627278] audit: type=1325 audit(1673777911.745:42): table=filter family=2 entries=8
[15371.718742] Module [xt_recent] is removed. 
[15371.803023] Module [xt_iprange] is removed. 
[15371.842365] Module [xt_limit] is removed. 
[15371.903011] Module [xt_state] is removed. 
[15371.962215] Module [xt_tcpudp] is removed. 
[15372.021926] Module [xt_multiport] is removed. 
[15372.096171] Module [xt_LOG] is removed. 
[15372.165659] Module [iptable_filter] is removed. 
[15408.899990] audit: type=1325 audit(1673777949.016:43): table=filter family=2 entries=0
[15409.193144] audit: type=1325 audit(1673777949.306:44): table=filter family=2 entries=4
[15409.216286] audit: type=1325 audit(1673777949.326:45): table=filter family=2 entries=8
[15421.548829] audit: type=1325 audit(1673777961.666:46): table=filter family=2 entries=9
[15421.572318] audit: type=1325 audit(1673777961.686:47): table=filter family=2 entries=8
[15421.621694] Module [xt_recent] is removed. 
[15421.711313] Module [xt_iprange] is removed. 
[15421.761077] Module [xt_limit] is removed. 
[15421.811011] Module [xt_state] is removed. 
[15421.861566] Module [xt_tcpudp] is removed. 
[15421.911008] Module [xt_multiport] is removed. 
[15421.985053] Module [xt_LOG] is removed. 
[15422.054470] Module [iptable_filter] is removed. 
[15456.219988] audit: type=1325 audit(1673777996.337:48): table=filter family=2 entries=0
[15456.391110] audit: type=1325 audit(1673777996.507:49): table=filter family=2 entries=4
[15456.414008] audit: type=1325 audit(1673777996.527:50): table=filter family=2 entries=8
[15463.757399] audit: type=1325 audit(1673778003.867:51): table=filter family=2 entries=9
[15463.780631] audit: type=1325 audit(1673778003.897:52): table=filter family=2 entries=8
[15463.829734] Module [xt_recent] is removed. 
[15463.920306] Module [xt_iprange] is removed. 
[15463.960264] Module [xt_limit] is removed. 
[15464.009937] Module [xt_state] is removed. 
[15464.060226] Module [xt_tcpudp] is removed. 
[15464.109830] Module [xt_multiport] is removed. 
[15464.183535] Module [xt_LOG] is removed. 
[15464.243327] Module [iptable_filter] is removed. 
[15782.695824] mii: exports duplicate symbol generic_mii_ioctl (owned by kernel)

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

eth0      Link encap:Ethernet  HWaddr 00:11:32:7E:F7:8A  
          inet addr:192.168.0.16  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::211:32ff:fe7e:f78a/64 Scope:Link
          inet6 addr: 240d:1a:2de:2e00:211:32ff:fe7e:f78a/64 Scope:Global
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:2667 errors:0 dropped:0 overruns:0 frame:0
          TX packets:2847 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:862825 (842.6 KiB)  TX bytes:3234509 (3.0 MiB)
          Interrupt:11 

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:68:00:32  
          inet addr:192.168.0.3  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: 240d:1a:2de:2e00:2e0:4cff:fe68:32/64 Scope:Global
          inet6 addr: fe80::2e0:4cff:fe68:32/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:49969 errors:0 dropped:0 overruns:0 frame:0
          TX packets:41424 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:11446812 (10.9 MiB)  TX bytes:38224400 (36.4 MiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:10257 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10257 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1411559 (1.3 MiB)  TX bytes:1411559 (1.3 MiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

powerpc970, 2023-01-15:
**Output of lsusb command**

|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb3          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.2.auto) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller xhci-hcd.5.auto) hub
  |__5-1         0bda:8156:3100 00  3.20 5000MBit/s 256mA 1IF  (Realtek USB 10/100/1G/2.5G LAN 001000001)

bb-qq, 2023-01-28:
From the output of `lsusb` it appears that the device is not recognized at the hardware level and driver loading is skipped.
Perhaps the USB port is not getting enough power.
A self-powered USB hub might solve this problem, but due to the limitations of the rtd1296 platform, it may be difficult to get it to work stably.

powerpc970, 2023-02-01:
Thanks for the reply.
I tried a self-powered USB hub. But unfortunately the symptoms were the same.
The r8152 driver started without the adapter connected.
If QNA-UC5G1T is not recognized, will the driver not start?

bb-qq, 2023-03-18:
> If QNA-UC5G1T is not recognized, will the driver not start?

Indeed, the driver package itself should boot. If you post the entire dmesg output here I might be able to find out something more about why the package is not starting.

But whatever it is, unless the adapter appears in the output of lsusb, it will not work even if the driver is loaded.

closed 2024-06-15 by bb-qq