bb-qq/aqc111 #102: Can't change between high and low power mode

johnjonjohn, 2023-03-06:
Running the command sudo 'echo on > /var/packages/aqc111/etc/flag-low_power_5g' brings up an error "command not found"

Changing the setting with ethtool does work but does not save upon rebooting the nas
( ethtool --set-priv-flags eth2 "Low Power 5G" on or off )

bb-qq, 2023-03-18:
Please try this command.  I will update the document.
```
echo on | sudo tee /var/packages/aqc111/etc/flag-low_power_5g
```

andmar197264, 2023-03-23:
Hello ! Please tell me what is the reason for the speed drop by more than 30% (((
Previously, the Qnap 5G adapter gave out more than 4Gb / s on the first versions of the drivers! Now more than 2.86Gb / s - it doesn’t rise! ((( I changed the troping disable parameters, it doesn’t help (

Thank you and thanks for your work for the people !!!!
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   170 MBytes  1.43 Gbits/sec                  
[  6]   0.00-1.00   sec   177 MBytes  1.48 Gbits/sec                  
[SUM]   0.00-1.00   sec   347 MBytes  2.91 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   1.00-2.00   sec   196 MBytes  1.64 Gbits/sec                  
[  6]   1.00-2.00   sec   167 MBytes  1.40 Gbits/sec                  
[SUM]   1.00-2.00   sec   363 MBytes  3.05 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   2.00-3.00   sec   161 MBytes  1.35 Gbits/sec                  
[  6]   2.00-3.00   sec   191 MBytes  1.60 Gbits/sec                  
[SUM]   2.00-3.00   sec   352 MBytes  2.95 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   3.00-4.00   sec   187 MBytes  1.57 Gbits/sec                  
[  6]   3.00-4.00   sec   180 MBytes  1.51 Gbits/sec                  
[SUM]   3.00-4.00   sec   367 MBytes  3.08 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   4.00-5.00   sec   203 MBytes  1.70 Gbits/sec                  
[  6]   4.00-5.00   sec   164 MBytes  1.37 Gbits/sec                  
[SUM]   4.00-5.00   sec   366 MBytes  3.07 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   5.00-6.00   sec   177 MBytes  1.48 Gbits/sec                  
[  6]   5.00-6.00   sec   186 MBytes  1.56 Gbits/sec                  
[SUM]   5.00-6.00   sec   362 MBytes  3.04 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   6.00-7.00   sec   182 MBytes  1.53 Gbits/sec                  
[  6]   6.00-7.00   sec   184 MBytes  1.54 Gbits/sec                  
[SUM]   6.00-7.00   sec   366 MBytes  3.07 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   7.00-8.00   sec   186 MBytes  1.56 Gbits/sec                  
[  6]   7.00-8.00   sec   173 MBytes  1.46 Gbits/sec                  
[SUM]   7.00-8.00   sec   360 MBytes  3.02 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   8.00-9.00   sec   187 MBytes  1.57 Gbits/sec                  
[  6]   8.00-9.00   sec   171 MBytes  1.43 Gbits/sec                  
[SUM]   8.00-9.00   sec   358 MBytes  3.00 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[  4]   9.00-10.00  sec   168 MBytes  1.41 Gbits/sec                  
[  6]   9.00-10.00  sec   180 MBytes  1.51 Gbits/sec                  
[SUM]   9.00-10.00  sec   348 MBytes  2.92 Gbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  1.77 GBytes  1.52 Gbits/sec                  sender
[  4]   0.00-10.00  sec  1.77 GBytes  1.52 Gbits/sec                  receiver
[  6]   0.00-10.00  sec  1.73 GBytes  1.49 Gbits/sec                  sender
[  6]   0.00-10.00  sec  1.73 GBytes  1.49 Gbits/sec                  receiver
[SUM]   0.00-10.00  sec  3.50 GBytes  3.01 Gbits/sec                  sender
[SUM]   0.00-10.00  sec  3.50 GBytes  3.01 Gbits/sec                  receiver

iperf Done.

johnjonjohn, 2023-03-23:
> Please try this command. I will update the document.
> 
> ```
> echo on | sudo tee /var/packages/aqc111/etc/flag-low_power_5g
> ```

Thank you, that seems to work.  Just to confirm, does that command enable low power mode?

Am I understanding correctly that
'echo on | sudo tee /var/packages/aqc111/etc/flag-low_power_5g'

Would enable low power mode and that...

'echo off | sudo tee /var/packages/aqc111/etc/flag-low_power_5g'

Would disable low power mode?

Thank you

bb-qq, 2023-04-22:
Yes, that is the correct understanding.
However, this operation only changes the flag to be read at driver startup, so a driver or system reboot is required to actually apply the settings to the device.

johnjonjohn, 2023-05-01:
> Yes, that is the correct understanding. However, this operation only changes the flag to be read at driver startup, so a driver or system reboot is required to actually apply the settings to the device.

Thanks.  All seems to be working well in both high power and low power.  I have ran for 3 weeks on low power and about a month on high power with no drop outs.

I have used Velcro to stick the nic to the back of the nas in line of the fan so that it is getting constant airflow.

bb-qq, 2023-05-13:
Glad to hear it is working well.

I will close this issue since you seem to have resolved your question, but I would appreciate it if you could write to #79 with information about the product and the fans you are using.

closed 2023-05-13 by bb-qq

johnjonjohn, 2023-05-13:
> Glad to hear it is working well.
> 
> I will close this issue since you seem to have resolved your question, but I would appreciate it if you could write to #79 with information about the product and the fans you are using.

Just a couple of days after writing this I have started having failures on high power mode.  I recently started seeding the Titanic for hour and glory demo and when the torrent gets busy the network will drop.  It only happens on high power.  Low power works fine in this same situation.