bb-qq/aqc111 #103: two usb network cards at the same time?

bg2vcj, 2023-03-10:
DS220+
Excuse me, can use two usb network cards at the same time?
What settings need to do?

bb-qq, 2023-03-18:
Currently, this driver does not handle the two interfaces well.

Also, even if the driver is modified, it may not work properly at the hardware level due to bandwidth issues of the USB host controller or power supply of the USB port.

bb-qq, 2024-06-15:
For Realtek adapters, it works in the following way I think the same method will work, so please try it. However, it may not have enough power.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#multiple-identical-devices-do-not-work

Since this issue is very old, so I'm closing it for now. Please feel free to re-open it if you have any questions.

closed 2024-06-15 by bb-qq