bb-qq/aqc111 #104: Repackage for DSM7.2 working for r8152

noblemd00, 2023-03-17:
Will a repackage for DSM 7.2 beta work? User almir1904 repackaged the r8152 (2.5 Gig) package for DSM 7.2 geminilake for his DS920 - I installed the package and after reboot it works fine on my DS1520+ (also geminilake) with a 2.5 USB adapter - not sure if that is all that is needed for this package. I would try doing what he did for the r8152, but I have no experience with Synology packages.

dtngx, 2023-03-18:
> Will a repackage for DSM 7.2 beta work? User almir1904 repackaged the r8152 (2.5 Gig) package for DSM 7.2 geminilake for his DS920 - I installed the package and after reboot it works fine on my DS1520+ (also geminilake) with a 2.5 USB adapter - not sure if that is all that is needed for this package. I would try doing what he did for the r8152, but I have no experience with Synology packages. If repackaged, I will test. All previous AQC111 releases worked perfectly (before 7.2b) on my DS1520 (after initial minor problems).

Hi noblemd00, i'm currently testing to get it running on my geminilake DS720+ I will keep you updated, if I manage to get it running.

dtngx, 2023-03-18:
I just got it to work by repackaging the driver. Basically it's just the modified Version of almir1904's step by step guide.

You need a Linux X86_64 environment and then

mkdir -p /toolkit
apt-get install cifs-utils python3 python3-pip git
git clone https://github.com/SynologyOpenSource/pkgscripts-ng
cd /toolkit/pkgscripts-ng/
git checkout DSM7.2
./EnvDeploy -v 7.2 -p geminilake (replace it with your CPU GEN)
mkdir source
cd source/
git clone https://github.com/bb-qq/aqc111.git

----
now modify the INFO.sh file to DSM 7.2 and 7.2-64216 requirements
Then:
----

cd /toolkit/pkgscripts-ng/
./PkgCreate.py -v 7.2 -p geminilake -c aqc111

After that, you can find two spk's in 
/toolkit/build-env/ds.(your CPU GEN)/image/packages/ 

I can provide a better walkthrough tonight ;)

noblemd00, 2023-03-18:
> I just got it to work by repackaging the driver. Basically it's just the modified Version of almir1904's step by step guide.
> 
Could you post your finished package?

dtngx, 2023-03-19:
try it out: 
https://github.com/dtngx/aqc111-beta/releases/download/beta/aqc111-geminilake-1.3.3.0-10.spk

noblemd00, 2023-03-19:
> Here is the repack, please let me know, if it works for you.
> 
> https://github.com/dtngx/aqc111-beta/releases/download/beta/aqc111-geminilake-1.3.3.0-10.spk

[image]

Thanks - Excellent. Connected to rear USB thru USB 3.1 hub - Speeds are approximately what I was getting on DSM7.1 and reconnects after restarting and after shutdown/startup (via MagicPacket WOL). Works with DHCP or static IP settings. 

8Tb x 5 RAID 5
‎SABRENT ‎NT-SS5G

dtngx, 2023-03-19:
Great, thanks for the feedback! What MTU did you set for the SABRENT NIC? Default 1500 or 9000 Jumbo Frames?

dtngx, 2023-03-21:
Very good, I'll try that. 9000 seems to be only marginally faster for me than 1500.

bb-qq, 2023-04-22:
I would like to close this issue since the driver for DSM 7.2 is already provided.
Thank you for all your help in getting things confirmed prior.

closed 2023-04-22 by bb-qq