bb-qq/aqc111 #105: DS3622xs+ (dsm 7.1.1 upd4) --> aqc111 loop error -

LeniReinfe, 2023-03-24:
### Description of the problem
doesn't repair 
> Replace this line with your answer.
after installation, and terminal bless , doesn't repair it

### Description of your products
NAS
> - the product name of your NAS model
>     - and output of ``uname -a`` command
DS3622xs+ (dsm 7.1.1 upd4) 
>     - DSM version x.x
> - the product name of the ethernet dongle

### Description of your environment

> - connect directly to a PC or using a hub
>     - PC: NIC model name or chipset name
NIC PC (1g) + NIC QNAP 10G
>     - Hub: model name
> - cable category (Cat.5e or Cat.6 or Cat.7)
6
### Output of ``dmesg`` command

````
(paste output here)
````

### Output of ``lsusb`` command

````
(paste output here)
````

### Output of ``ifconfig -a`` command

````
(paste output here)
````

bb-qq, 2023-04-22:
Could you paste the content of `/var/packages/aqc111/var/log/start-stop-status.log`?

bb-qq, 2024-06-15:
If you have difficulty submitting logs, please try a complete reinstallation.
https://github.com/bb-qq/r8152/wiki/Troubleshooting#uninstall-completely--full-reinstallation

closed 2024-06-15 by bb-qq