bb-qq/aqc111 #107: Using both 2.5G (CLUB 3D CAC-1420) and the 5G (QNA-UC5G1T or NT-SS5G) Adapter

dedura, 2023-03-29:
### Description of the problem

> I would like to use both 2.5G (CLUB 3D CAC-1420) and the 5G (QNA-UC5G1T or NT-SS5G) Adapter at the same time.
I have installed both drivers "aqc111" and "r8152".
Both LAN connections come up as LAN3 and LAN4 and I can configure the IP addresses for both.
However, upon reboot, the IPs change (169.x.x.x) for both and I have to reconfigure.
Also, when using both connections at the same time, either LAN3 or LAN4 disconnects and I have to reload the driver.
Is there any fix for this, can we use both adapters at the same time?

### Description of your products

NAS: Synology DS918+
DSM: 7.1.1-42962 Update 4
Adapter: SABRENT NT-SS5G (Front Port)
Driver: 1.3.3.0-10 DSM-7.x (reuploaded)
RAM: 8GB
RearUSB Port used for: 
 - SABRENT 4 Port USB 3.0 Hub
1. (UPS) CP1500EPFCLCD - Cyber Power System, Inc.
2. External Hard Drive for Backup (Switched off)
3. 2.5G Club 3D CAC-1420

### Description of your environment

1. Connection: From "DS918+" to PC's NIC "Marvell AQtion AQC107 10Gb Ethernet" using (NT-SS5G)
2. Connection: From"DS918+" to Intel NUC "Intel 1225-V" using (CLUB 3D CAC-1420)

hy-ts, 2023-04-09:
> However, upon reboot, the IPs change (169.x.x.x) for both and I have to reconfigure.

I have partly similar problem in case of r8152 driver: [https://github.com/bb-qq/r8152/issues/300](url)
You could try to make in DSM Task Scheduler a new task (named e.g.: Set LAN 3 and LAN 4):
User: root; Event: System boot; User defined script (simply write these in):
sudo ip link set eth2 up
/usr/syno/sbin/synonet --manual eth2 192.168.0.11 255.255.255.0 (change to your network address and subnet mask for LAN 3)
sudo ip link set eth3 up
/usr/syno/sbin/synonet --manual eth3 192.168.0.12 255.255.255.0 (change to your network address and subnet mask for LAN 4)

This "script" starts the LAN 3 (eth2) and LAN 4 (eth3), and sets the IP address and subnet mask at boot.
The "sudo ip link set ..." is not necessary in your case, it only turns on/up the network port, to be sure.
For me these worked in case of two Club 3D CAC-1420 USB adapters in DS918+.

> Is there any fix for this, can we use both adapters at the same time?

I'm not sure, but the two USB ports on the DS918+ might be capable of a total of 5Gbps. And if that's the case, couldn't your 5Gbps and 2.5Gbps network adapters be overloading the NAS if they're running at the same time? Maybe that's why the two 2.5Gbps network adapters working for me at the same time.

dedura, 2023-04-10:
@hy-ts  - Thank you very much for your feedback. I tried this and it receives the provided IP addresses, but then I run into an issue when using both Adapters simultaneously, only transferring a few Megabytes, one of the LANs just randomly stops and disappears from the Network Settings. I then have to restart the driver manually.
I have not maxed them out, it just doesn't work properly when transferring at the same time.

Example:
PC to NAS is constantly sending some bytes to stay alive with the UPS software, CCTV or Synology Drive.
When transferring from Intel NUC to the NAS with the 2.5G Adapter, one of the drivers just stopped.

hy-ts, 2023-04-10:
If the 2.5Gbps adapter is unplugged from the DS918+ and you only use the 5Gbps adapter, does the 5Gbps adapter work stably? So, even without the Task Manager script, it gets the correct IP address/Netmask automatically after a reboot, and this one adapter doesn't randomly disconnect even after several days of use?

dedura, 2023-04-10:
Hi, yes that's correct, if the 2.5G is unplugged, the 5Gbps adapter works without any issues and is stable and no, I don't have to manually run a script via task manager.
I recently had issues with freezing, crashes and whole NAS reboot - the solution for me was to replace my 16GB RAM with 2x 4GB on my DS918+. I also was using the M.2 SSDs as Storage Volume, unmounting that - all issues were gone and the 5Gbps Adapter works stable.

hy-ts, 2023-04-10:
My guess is that you won't be able to get more data transfer speeds out of the two USB 3 connectors than a total of 5Gbps, in the case of the DS918+. "bb-qq" also wrote about this, as I quoted above ("... bandwidth issues of the USB host controller ..."). In the case of a newer, different Synology NAS, where each USB connector is capable of 5Gbps data transfer bandwidth by itself, your 5Gbps and 2.5Gbps adapters would probably work at the same time. (I don't know which NAS is like that.) 

Unfortunately, the DS918+ has several hardware limitations (for example, the two NVMe SSD connectors together have one PCIe 2.0 speed connection, with a maximum speed of 500 MByte/s).

Sorry to say, but I have to recommend that you only use the 5Gbps network adapter and remove the 2.5Gbps network adapter.
Then if you switch to DSM 7.2, you can try to see if they work at the same time, but I don't think they will work at the same time either (due to USB hardware limitation).

Comment about NAS reboots:
I think too much hardware in the NAS may cause power supply problems. The DS918+ has a 100W power supply.
Let's add up the maximum, even short-term consumption of each hardware element.
CPU: 10W
Mainboard: 15W (approx.)
4 HDD: 4x10W=40W
2 NVMe SSDs (2x1TB): 2x7W=14W
16GB RAM: 8W
USB 3.0 port: 5Volt x 0.9Amp=4.5W
Total: 91.5W
In case of longer file copying, when the CPU is loaded to 100% with some other services (say, video transcoding), this consumption close to 100W will be for a long time, and this is certainly not about stability, because of the 100W power supply.

dedura, 2023-04-11:
@hy-ts - thanks a lot for your input. Yes I have removed the 2.5G Adapter and only using the 5G Adapter.

hy-ts, 2023-04-11:
**@dedura** - I wish stable operation for your tuned DS918+!

bb-qq, 2023-04-22:
I agree that this could cause power consumption issues. Using a self-powered hub may work around the problem, but as mentioned, this time there is concern about transfer bandwidth.

bb-qq, 2024-06-15:
I think we do not have unfinished action with this issue, so I would like to close it.
Please feel free to re-open the issue.

closed 2024-06-15 by bb-qq