bb-qq/aqc111 #109: DS920+ Using AQC111 r 1.3.3.0-11 for DSM 7.2 beta with QNAP-UC5G1T - repeated crashes under load

HCATmike, 2023-04-28:
### Description of the problem

> DS920+ Using AQC111 r 1.3.3.0-11 for DSM 7.2 beta with QNAP-UC5G1T - repeated crashes under load. Will not run for more than a day or two. If I stop, and restart in package center, resumes working just fine. W/R performance around 250 MB/sec when working - per Blackmagic 

### Description of your products

>    -Synology (S920+
>     - Linux BartNAS 4.4.302+ #64216 SMP Tue Mar 7 17:09:39 CST 2023 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM version 7.2-64216 beta
>      -QNA-UC5G1T

### Description of your environment

>     - HUB 
>     - iMac 20'1  with Intel 2.5 Gb Killer networking E3100
>     - Trendmicro TEG-S380
>     - cable category Cat.5e

### Output of ``dmesg`` command

````

[477919.796995] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477919.803726] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477919.810444] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477919.817158] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477928.283140] net_ratelimit: 36 callbacks suppressed
[477928.288593] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477929.295424] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477929.335056] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.485025] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.491751] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.676410] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.683155] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.689880] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.696601] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477930.703322] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.301243] net_ratelimit: 2158 callbacks suppressed
[477934.306894] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.313615] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.320336] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.333401] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.379493] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.386228] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.392965] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.399697] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.406435] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477934.413156] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.301077] net_ratelimit: 523713 callbacks suppressed
[477939.306935] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.314092] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.321088] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.327820] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.334555] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.342088] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.349081] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.355818] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.362550] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477939.370083] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.459490] net_ratelimit: 1021332 callbacks suppressed
[477944.465432] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.472157] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.478871] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.486418] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.493151] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.502169] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.508901] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.515624] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.524463] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[477944.531193] aqc111 2-2:1.0 eth2: kevent 2 may have been dropped
[... 493 lines cut ...]
eth2      Link encap:Ethernet  HWaddr 24:5E:BE:7B:C3:1D  
          inet addr:192.168.1.225  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe7b:c31d/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:24968313 errors:1 dropped:0 overruns:0 frame:0
          TX packets:4779673 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:30201950577 (28.1 GiB)  TX bytes:28838235112 (26.8 GiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:4177013 errors:0 dropped:0 overruns:0 frame:0
          TX packets:4177013 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:1706942028 (1.5 GiB)  TX bytes:1706942028 (1.5 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

HCATmike, 2023-04-29:
Appears that load makes failure of driver more frequent, but it does happen eventually when not under much load. The DS920+ is used in a home, only by me. I just did a reboot, and here are the Blackmagic results- 
[image]

bb-qq, 2023-04-30:
Was it a bit more stable before DSM 7.1? Or is this the first time you've run it with DSM 7.2?

Also, have you looked at this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting

HCATmike, 2023-04-30:
Thank you for your reply. This is the first time I have run it, honestly. My DS 920+ already had the RC on it when I installed your driver. It is so fantastic, when it works, but I understand this may be a DSM RC problem. I reported it for the communities awareness, and to see if any other are having similar issues with the RC.Are other users seeing dropped connections with the DSM RC?

I somehow missed the trouble shooting page, so I will try some of the options I see there and report back if I find something that helps.

Thank you again,

Michael

parko86, 2023-06-30:
How is this working in DSM 7.2 now? Thanks!

bb-qq, 2023-07-15:
The official version of DSM 7.2 has already been released and is reported to work.

Since there have been no posts for a long period of time and there does not seem to be anything left to do in this Issue, I would like to close this Issue.

closed 2023-07-15 by bb-qq