bb-qq/aqc111 #110: No Jumbo Frames settings on DS423+ with Sabrent NT-SS5G

BioTeq, 2023-05-02:
### Description of the problem

There is no option to set the jumbo frames in the network interface. Dropdown shows only up to the default 1500.
The interface is reporting correctly 5gbe, the netgear switch is recognizing the 5gbe connection and is by default jumbo frame enabled.
Initially I've installed the [1.3.3.0-11](https://github.com/bb-qq/aqc111/releases/tag/1.3.3.0-11) and then realized after reddit chat that I was on a pre-release.
After uninstalling I've installed the 1.3.3.0-10, no change - same issue.
I've upgraded again to -11, no difference.
Lan Cable on the rear of the Synology was tested while plugged in and taken out, no difference.

### Description of your products

DS423+
Linux Synology 4.4.180+ #42962 SMP Sat Apr 8 00:32:22 CST 2023 x86_64 GNU/Linux synology_geminilake_423+
DSM 7.1.1-42962 Update 5
Sabrent NT-SS5G connected to the front USB Port (USB 3.2)

### Description of your environment

Switch Netgear GS110EMX
PC... probably irrelevant - Mac with an OWC Thunderbolt 3 10G (OWCTB3ADP10GBE)
I'm getting 10gbe on this interface with my DS1821+
Cables: Cat7 and Cat8 tested.

### Output of ``dmesg`` command

````
[166785.737760] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[166884.447941] mii: module verification failed: signature and/or required key missing - tainting kernel
[166884.930779] usbcore: registered new interface driver aqc111
[166884.942712] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-1, Aquantia AQtion USB to 5GbE Controller, 00:24:27:88:46:81
[166885.279825] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[166892.735391] audit: type=1325 audit(1682877216.655:411): table=filter family=2 entries=33
[166892.749832] audit: type=1325 audit(1682877216.669:412): table=filter family=2 entries=33
[166892.796737] audit: type=1325 audit(1682877216.716:413): table=filter family=2 entries=33
[166892.811552] audit: type=1325 audit(1682877216.731:414): table=filter family=10 entries=32
[166892.827374] audit: type=1325 audit(1682877216.747:415): table=filter family=10 entries=32
[166892.842424] audit: type=1325 audit(1682877216.762:416): table=filter family=10 entries=32
[166892.857282] audit: type=1325 audit(1682877216.777:417): table=filter family=2 entries=37
[166892.872107] audit: type=1325 audit(1682877216.792:418): table=filter family=10 entries=36
[166892.886887] audit: type=1325 audit(1682877216.806:419): table=filter family=2 entries=30
[166892.901621] audit: type=1325 audit(1682877216.821:420): table=filter family=10 entries=22
[167080.614774] aqc111 2-1:1.0 eth2: Link Speed 5000, USB 3
[167080.627810] IPv6: ADDRCONF(NETDEV_CHANGE): eth2: link becomes ready
[167082.475657] audit_printk_skb: 108 callbacks suppressed
[167082.481501] audit: type=1325 audit(1682877406.433:457): table=filter family=2 entries=33
[167082.496690] audit: type=1325 audit(1682877406.454:458): table=filter family=2 entries=34
[167083.786745] audit: type=1325 audit(1682877407.744:459): table=filter family=2 entries=35
[167083.802178] audit: type=1325 audit(1682877407.759:460): table=filter family=2 entries=34
[167086.527134] audit: type=1325 audit(1682877410.485:461): table=filter family=2 entries=33
[167086.793788] audit: type=1325 audit(1682877410.752:462): table=filter family=2 entries=33
[167086.809330] audit: type=1325 audit(1682877410.767:463): table=filter family=2 entries=34
[167087.541199] audit: type=1325 audit(1682877411.499:464): table=filter family=2 entries=35
[167087.556499] audit: type=1325 audit(1682877411.514:465): table=filter family=2 entries=35
[167087.571288] audit: type=1325 audit(1682877411.529:466): table=filter family=10 entries=32
[167087.585973] audit: type=1325 audit(1682877411.544:467): table=filter family=10 entries=32
[167087.601233] audit: type=1325 audit(1682877411.559:468): table=filter family=10 entries=32
[167087.616559] audit: type=1325 audit(1682877411.575:469): table=filter family=2 entries=39
[167087.631148] audit: type=1325 audit(1682877411.589:470): table=filter family=10 entries=36
[167087.645940] audit: type=1325 audit(1682877411.604:471): table=filter family=2 entries=30
[167087.660643] audit: type=1325 audit(1682877411.619:472): table=filter family=10 entries=22
[167087.702731] audit: type=1325 audit(1682877411.661:473): table=filter family=2 entries=23
[174215.579153] audit_printk_skb: 105 callbacks suppressed
[174215.585001] audit: type=1325 audit(1682884540.945:509): table=filter family=2 entries=33
[174215.600481] audit: type=1325 audit(1682884540.967:510): table=filter family=2 entries=34
[174218.250299] audit: type=1325 audit(1682884543.617:511): table=filter family=2 entries=35
[174218.264424] audit: type=1325 audit(1682884543.631:512): table=filter family=2 entries=35
[174218.303704] audit: type=1325 audit(1682884543.670:513): table=filter family=2 entries=35
[174218.319226] audit: type=1325 audit(1682884543.686:514): table=filter family=10 entries=32
[174218.333912] audit: type=1325 audit(1682884543.701:515): table=filter family=10 entries=32
[174218.348939] audit: type=1325 audit(1682884543.716:516): table=filter family=10 entries=32
[174218.364053] audit: type=1325 audit(1682884543.731:517): table=filter family=2 entries=39
[174218.380656] audit: type=1325 audit(1682884543.747:518): table=filter family=10 entries=36
[174220.909538] aqc111 2-1:1.0 eth2: Link Speed 5000, USB 3
[174226.133900] audit_printk_skb: 114 callbacks suppressed
[174226.139762] audit: type=1325 audit(1682884551.502:557): table=filter family=2 entries=33
[174226.154061] audit: type=1325 audit(1682884551.522:558): table=filter family=2 entries=33
[174226.168709] audit: type=1325 audit(1682884551.537:559): table=filter family=2 entries=33
[174226.182948] audit: type=1325 audit(1682884551.551:560): table=filter family=10 entries=32
[174226.197474] audit: type=1325 audit(1682884551.566:561): table=filter family=10 entries=32
[174226.212312] audit: type=1325 audit(1682884551.581:562): table=filter family=10 entries=32
[174226.227142] audit: type=1325 audit(1682884551.595:563): table=filter family=2 entries=37
[174226.241729] audit: type=1325 audit(1682884551.610:564): table=filter family=10 entries=36
[174226.256483] audit: type=1325 audit(1682884551.625:565): table=filter family=2 entries=30
[174226.270990] audit: type=1325 audit(1682884551.639:566): table=filter family=10 entries=22
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65002A7E2E346E97)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.180+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         2eca:c101:0101 00  3.20 5000MBit/s 896mA 1IF  (Aquantia Aquantia 5G USB Ethernet Adapter 0000001k)
````

### Output of ``ifconfig -a`` command

````
eth2      Link encap:Ethernet  HWaddr 00:24:27:88:46:81  
          inet addr:192.168.3.125  Bcast:192.168.3.255  Mask:255.255.255.0
          inet6 addr: ... Scope:Link #removed address
          inet6 addr: ... Scope:Global #removed address
          inet6 addr: ... Scope:Global #removed address
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:93173 errors:0 dropped:0 overruns:0 frame:0
          TX packets:33341 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:15578613 (14.8 MiB)  TX bytes:7161048 (6.8 MiB)
````

BioTeq, 2023-05-02:
PS. The interface speed tested with iperf3 computer to ds423+ is fluctuating between 1gbps and ~2.5gbps. 
The Sabrent adapter itslf tops out at about 3.5gbps when tested on a computer against the DS1821+ 10Gbe NIC (as expected with this chip and USB interface)

BioTeq, 2023-05-02:
PS2. Setting the interface manually to MTU 9000 (ip link set eth2 mtu 9000) completely kills the speed - it's down to about 0.1 to 0.2gbps

bb-qq, 2023-05-13:
Is the situation the same after rebooting DS423+?

BioTeq, 2023-05-19:
Sorry for the late reply, but I couldn't shut down the server in the middle of a rebuild. 
Tested, yes - still showing 1500 after a reboot.
Now I also had a random disconnect, after about 2 weeks of operation - the interface was showing that it's connected and has an ip assigned, but effectively no connection. I had to shutdown and restart the driver to regain access.

BioTeq, 2023-05-19:
Here is also the dmesg output trimmed a bit:

`f0
[625797.257921] xhci_hcd 0000:00:15.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 5 comp_code 1
[625797.269485] xhci_hcd 0000:00:15.0: Looking for event-dma 0000000068330ff0 trb-start 0000000068a80fa0 trb-end 0000000000000000 seg-start 0000000068a80000 seg-end 0000000068a80ff0
[625797.287257] xhci_hcd 0000:00:15.0: Looking for event-dma 0000000068330ff0 trb-start 0000000068330000 trb-end 00000000683300c0 seg-start 0000000068330000 seg-end 0000000068330ff0

... many many many of those repeated

[625802.735066] 4RPC:  2040 call_timeout in call status stage
[625802.741299] nfs: server 192.168.3.12 not responding, timed out
[625806.798242] 4RPC:  2041 call_timeout in call status stage
[625806.804485] nfs: server 192.168.3.12 not responding, timed out
[625806.877196] 4RPC:  2042 call_timeout in call status stage
[625806.883440] nfs: server 192.168.3.12 not responding, timed out
[625807.066171] 4RPC:  2043 call_timeout in call status stage
[625807.072430] nfs: server 192.168.3.12 not responding, timed out
[625807.949987] ------------[ cut here ]------------
[625807.955262] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:307 dev_watchdog+0x214/0x220()
[625807.964794] NETDEV WATCHDOG: eth2 (aqc111): transmit queue 0 timed out
[625807.972187] Modules linked in: nfsv3 nfs nfnetlink xfrm_user xfrm_algo xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_addrtype xt_REDIRECT xt_nat nf_nat_redirect vhost_scsi(O) vhost(O) tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) aqc111(OE) usbnet(OE) mii(OE) fuse ipt_MASQUERADE nf_nat_masquerade_ipv4 vfat fat tun udf isofs iptable_nat nf_nat_ipv4 nf_nat synoacl_vfs(PO) xt_conntrack xt_mac xt_geoip(O) compat_xtables(O) nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_filter ip_tables
[625808.051613]  x_tables raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx 8021q nfsd btrfs ecryptfs zstd_decompress zstd_compress xxhash xor raid6_pq lockd grace rpcsec_gss_krb5 auth_rpcgss sunrpc uas usb_storage leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper syno_hddmon(P) geminilake_synobios(PO) hid_generic usbhid hid usblp i915 drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops cfbfillrect cfbcopyarea cfbimgblt drm drm_panel_orientation_quirks iosf_mbi fb fbdev video backlight button i2c_algo_bit uhci_hcd ehci_pci ehci_hcd r8168(O) dm_crypt sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd ecb aes_x86_64 authenc des_generic
[625808.129387]  ansi_cprng cts md5 cbc cpufreq_powersave cpufreq_performance acpi_cpufreq processor cpufreq_stats vxlan zram ip6_udp_tunnel udp_tunnel ip_tunnel loop sha256_generic synorbd(PO) synofsbd(PO) etxhci_hcd xhci_pci xhci_hcd usbcore usb_common [last unloaded: rodsp_ep]
[625808.155340] CPU: 0 PID: 0 Comm: swapper/0 Tainted: P           OE   4.4.302+ #64551
[625808.163989] Hardware name: Synology DS423+/DS423+, BIOS M.901.00 06/13/2022
[625808.171859]  0000000000000000 ffff8804ffc03dd0 ffffffff8156f96f ffff8804ffc03e18
[625808.180224]  ffffffff817731ca ffff8804ffc03e08 ffffffff8104ca71 0000000000000000
[625808.188589]  ffff8804deca2000 0000000000000000 0000000000000001 ffff8804ffc16300
[625808.196957] Call Trace:
[625808.199780]  <IRQ>  [<ffffffff8156f96f>] dump_stack+0x57/0x6d
[625808.206305]  [<ffffffff8104ca71>] warn_slowpath_common+0x81/0xd0
[625808.213110]  [<ffffffff8104cb07>] warn_slowpath_fmt+0x47/0x50
[625808.219625]  [<ffffffff814dd2c4>] dev_watchdog+0x214/0x220
[625808.225847]  [<ffffffff814dd0b0>] ? netif_carrier_off+0x30/0x30
[625808.232558]  [<ffffffff810a20c5>] call_timer_fn+0x35/0x140
[625808.238780]  [<ffffffff814dd0b0>] ? netif_carrier_off+0x30/0x30
[625808.245489]  [<ffffffff810a3fd4>] run_timer_softirq+0x174/0x370
[625808.252196]  [<ffffffff81050e2d>] __do_softirq+0xdd/0x330
[625808.258323]  [<ffffffff810511b3>] irq_exit+0x73/0xa0
[625808.263962]  [<ffffffff81031bf3>] smp_apic_timer_interrupt+0x43/0x60
[625808.271155]  [<ffffffff81584fdb>] apic_timer_interrupt+0x9b/0xa0
[625808.277956]  <EOI>  [<ffffffff8147f305>] ? cpuidle_enter_state+0x135/0x2b0
[625808.285738]  [<ffffffff8147f4b2>] cpuidle_enter+0x12/0x20
[625808.291863]  [<ffffffff810858ef>] cpu_startup_entry+0x29f/0x320
[625808.298570]  [<ffffffff8157f274>] rest_init+0x72/0x74
[625808.304307]  [<ffffffff818f7d80>] start_kernel+0x3d2/0x3de
[625808.310531]  [<ffffffff818f7284>] x86_64_start_reservations+0x29/0x2b
[625808.317817]  [<ffffffff818f7369>] x86_64_start_kernel+0xe3/0xe6
[625808.324567] ---[ end trace 078f430d5bb23bfb ]---
[625821.834925] 4RPC:  2044 call_timeout in call status stage
[625821.841176] nfs: server 192.168.3.12 not responding, timed out
[625821.886982] 4RPC:  2045 call_timeout in call status stage
[625821.893233] nfs: server 192.168.3.12 not responding, timed out
[625822.076962] 4RPC:  2046 call_timeout in call status stage
[625822.083222] nfs: server 192.168.3.12 not responding, timed out
[625836.871644] 4RPC:  2047 call_timeout in call status stage

... many of those repeated

[... cut ...]
[642507.790299] audit: type=1325 audit(1684469984.396:1037): table=filter family=2 entries=35
[642507.804978] audit: type=1325 audit(1684469984.410:1038): table=filter family=2 entries=34
[642512.221351] 4RPC:  2273 call_timeout in connect stage
[642512.227201] nfs: server 192.168.3.12 not responding, timed out
[642512.233873] 4RPC:  2274 call_timeout in connect stage
[642512.239723] nfs: server 192.168.3.12 not responding, timed out
[642814.747552] 4RPC:  2275 call_timeout in connect stage
[642814.753401] nfs: server 192.168.3.12 not responding, timed out`

BioTeq, 2023-05-19:
PS. This seems weird as well...

ethtool -g eth2
Ring parameters for eth2:
Cannot get device ring settings: Operation not supported

BioTeq, 2023-06-03:
After further testing I've experienced a complete interface crash. I don't know what happened, it just disappeared. I couldn't bring it back to live, disconnecting, reconnecting nothing worked. Eventually I've installed the final version of 7.2, and attempted to reinstall the driver. After 4-5 uninstall/reinstall rounds it finally came back. The USB device was showing all the time in the Info Page, just the interface wouldn't show up in the Network page of the Control Panel.
No idea what's going on. It's been up for about a week now.

Another issue - although iPerf was correctly showing the speeds at a little over 3gbit, the actual data transfer rates as tested with Blackmagic Disk Speed Test were just awful (~40MB/s). Troubleshooting page helped here - the interface is reporting the wrong link speed, although all the settings are correct. I had to use the posted command:
`ethtool -s eth2 speed 5000 duplex full autoneg off `
So now I have to run two commands, this one, plus the MTU.
Working OK at the moment, would be great if the developers could have a look and fix on their end.

bb-qq, 2023-07-02:
Thanks for the report.

So far I have no idea why the jumbo frame settings are not showing up. I also don't know why auto-negotiation doesn't work, but this one is sometimes reported, so it is probably a hardware interoperability issue.

BioTeq, 2023-07-02:
Gotcha, I've been stable for about a month now running on DSM 7.2-64570 and I've upgraded to DSM 7.2-64570-1 yesterday (pretty important changelog), but the interface came back up flawlessly.

bb-qq, 2024-06-15:
I would like to close this issue as the manual configuration seems to have solved the problem.

closed 2024-06-15 by bb-qq