bb-qq/aqc111 #116: Linux 6.2 support

mattrpav, 2023-08-07:
### Description of the problem

Add support for building the driver with Linux 6.2 kernels

### Description of your products

> - Sabrent nt-ss5g

bb-qq, 2023-08-11:
What is the intent of this request, is there a model with Linux 6.2?

mattrpav, 2023-08-14:
Looking to compile from source to run Sabrent 5Gb USB adapter w/ Linux Kernel 6.2.x on Ubuntu 22.04

mattrpav, 2023-08-14:
Compile errors
```
In function ‘aqc111_set_mac_addr’:
.../aqc111-1.3.3.0-12/aqc111.c:737:46: warning: passing argument 6 of ‘aqc111_write_cmd’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
  737 |                                 ETH_ALEN, net->dev_addr);
      |                                           ~~~^~~~~~~~~~

continues.. 2 errors, 6 warnings total
```

bb-qq, 2023-10-09:
This driver is intended to be built using Synology Toolkit. Other build environments are not planned to be ported at this time.

If you want to run the aqc111 driver on a PC, use the driver that comes with the Linux kernel.

closed 2023-10-09 by bb-qq