bb-qq/aqc111 #117: Only Gigabit Speeds with DSM 7.2-64570 and DSM620 Slim (Driver v1.3.3.0-12)

ModeSevenIndustrialSolutions, 2023-08-07:
### Description of the problem

DSM version 7.2-64570 and only getting 1G interface speed with Netgear MS510TX Switch
(think this used to work at 5G with older system software and aqc111 driver)

$ uname -a
Linux SynologyNAS 4.4.302+ #64570 SMP Tue Jun 13 21:40:17 CST 2023 x86_64 GNU/Linux synology_apollolake_620slim

### Description of your products

Synology DSM620 Slim

### Description of your environment

Cat7 cable; cable has been checked/replaced and no improvement

### Output of ``dmesg`` command

[   89.597328] usbcore: registered new interface driver aqc111
[   89.609225] aqc111 1-3:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-3, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:7e:ca:81
[   90.084909] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[   90.272908] device ovs_eth2 entered promiscuous mode
[   90.278483] audit_printk_skb: 6 callbacks suppressed
[   90.284054] audit: type=1700 audit(1691423183.036:18): dev=ovs_eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[   90.404222] device eth2 entered promiscuous mode
[   90.409412] audit: type=1700 audit(1691423183.167:19): dev=eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[   90.477798] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[   96.177536] audit: type=1325 audit(1691423188.936:20): table=nat family=2 entries=0
[   96.182600] aqc111 1-3:1.0 eth2: Link Speed 1000, USB 2
[   96.199030] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready

### Output of ``lsusb`` command

$ lsusb
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-3         1c04:0015:0101 00  2.10  480MBit/s 500mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 2BI03114)
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 65001891E2345A95)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

### Output of ``ifconfig -a`` command

$ ifconfig -a
eth0      Link encap:Ethernet  HWaddr 90:09:D0:06:55:B3  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:790 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3273 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:113583 (110.9 KiB)  TX bytes:3991072 (3.8 MiB)
          Interrupt:115 base 0x2000  

eth1      Link encap:Ethernet  HWaddr 90:09:D0:06:55:B4  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:6328 errors:0 dropped:0 overruns:0 frame:0
          TX packets:3138 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5321450 (5.0 MiB)  TX bytes:3417921 (3.2 MiB)
          Interrupt:116 base 0x2000  

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:7E:CA:81  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:271 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:56586 (55.2 KiB)  TX bytes:6768 (6.6 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:187 errors:0 dropped:0 overruns:0 frame:0
          TX packets:187 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:21808 (21.2 KiB)  TX bytes:21808 (21.2 KiB)

ovs-syste Link encap:Ethernet  HWaddr 2A:C9:D7:8D:58:51  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_bond0 Link encap:Ethernet  HWaddr 90:09:D0:06:55:B3  
          inet addr:192.168.1.6  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::9209:d0ff:fe06:55b3/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:5100 errors:0 dropped:0 overruns:0 frame:0
          TX packets:5471 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:5228687 (4.9 MiB)  TX bytes:7408443 (7.0 MiB)

ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:7E:CA:81  
          inet addr:192.168.1.2  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fdea:6365:627b:4a01:265e:beff:fe7e:ca81/64 Scope:Global
          inet6 addr: fe80::265e:beff:fe7e:ca81/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:248 errors:0 dropped:0 overruns:0 frame:0
          TX packets:44 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:55264 (53.9 KiB)  TX bytes:6201 (6.0 KiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

### Output of ``ethtool eth2`` command

$ ethtool eth2
Settings for eth2:
	Supported ports: [ TP MII ]
	Supported link modes:   100baseT/Full 
	                        1000baseT/Full 
	Supported pause frame use: No
	Supports auto-negotiation: Yes
	Supported FEC modes: Not reported
	Advertised link modes:  100baseT/Full 
	                        1000baseT/Full 
	Advertised pause frame use: No
	Advertised auto-negotiation: Yes
	Advertised FEC modes: Not reported
	Speed: 1000Mb/s
	Duplex: Full
	Port: MII
	PHYAD: 0
	Transceiver: internal
	Auto-negotiation: on
Cannot get wake-on-lan settings: Operation not permitted
	Current message level: 0x00000007 (7)
			       drv probe link
	Link detected: yes

bb-qq, 2023-08-11:
```
|__1-3 1c04:0015:0101 00 2.10 480MBit/s 500mA 1IF (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 2BI03114)
```

This line indicates that the device is operating with a USB 2.0 connection.
Try changing the connection port or replace the USB cable.

ModeSevenIndustrialSolutions, 2023-08-15:
Wow, I'd tried two different cables and thought I had already checked this, but you were right about the cabling. Thanks, this is working again at 5G now.

closed 2023-08-15 by ModeSevenIndustrialSolutions