bb-qq/aqc111 #120: aqc111-geminilake-1.3.3.0-11_7.2.spk 权限问题望修复

liuyongchao, 2023-09-11:
```
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/rm -rf /var/packages/aqc111 ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/mkdir -p /var/packages/aqc111 -m 755
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/mkdir -p /var/packages/aqc111 -m 755 ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/touch /var/packages/aqc111/installing
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/touch /var/packages/aqc111/installing ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/INFO /var/packages/aqc111/INFO
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/INFO /var/packages/aqc111/INFO ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/rm -rf /var/packages/aqc111/scripts
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/rm -rf /var/packages/aqc111/scripts ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/scripts /var/packages/aqc111/scripts
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/scripts /var/packages/aqc111/scripts ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/rm -rf /var/packages/aqc111/conf
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/rm -rf /var/packages/aqc111/conf ret=[0]
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 Begin /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/conf /var/packages/aqc111/conf
2023/09/11 23:56:30     install aqc111 1.3.3.0-12 End /bin/mv -f /volume1/@tmp/synopkg/install.zS2jfh/conf /var/packages/aqc111/conf ret=[0]
2023/09/11 23:56:31     install aqc111 1.3.3.0-12 Begin postinst
/var/packages/aqc111/scripts/postinst: line 13: /dev/stderr: Permission denied
2023/09/11 23:56:31     install aqc111 1.3.3.0-12 End postinst ret=[5]
```
```
DS720+ 
aqc111-geminilake-1.3.3.0-11_7.2.spk
```

bb-qq, 2023-10-09:
Are you installing according to the instructions in the README?

liuyongchao, 2023-10-14:
> Are you installing according to the instructions in the README?

naxxm@Ds720:/$ sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su
install: cannot stat '/var/packages/aqc111/target/aqc111/spk_su': No such file or directory
naxxm@Ds720:/$
提示命令执行出错

closed 2023-10-22 by liuyongchao