bb-qq/aqc111 #121: DS920+ cannot up the link, with GUI displaying `DISCONNECTED` on QNA-UC5G1T

udmada, 2023-09-21:
### Description of the problem
- The interface `LAN3` is shown `DISCONNECTED` on the Web GUI
- The interface `LAN3` is only shown when ovs is enabled
- Manual `ifconfig ovs_eth2 up` exits 0
- Manual `ifconfig eth2 up` exits 0
- By default the newly created `LAN3` inteface uses DHCP but it doesn't communicate with the router/gateway/dns
- manually set up static IP/DNS for interface `LAN3`
- routing is still missing
- there is one line `mii: module verification failed: signature and/or required key missing - tainting kernel` in the logs not sure if related

#### Self-conducted trouble-shooting
- [x] followed the manual
- [x] reinstalled the driver multiple times
- [x] reboot multiple times
- [x] changed usb dongle cable
- [x] used rear usb port
- [x] used front usb port
- [x] used different Cat6 Cat6A cables
- [x] connected to different ports on the router

### Description of your products

- `uname -a`: Linux Lil-Nas 4.4.302+ #64570 SMP Tue Jun 13 21:39:55 CST 2023 x86_64 GNU/Linux synology_geminilake_920+
- DSM version: DSM 7.2-64570 Update 1
- Adaptor: QNAP QNA-UC5G1T
- Connection: Cat6A directly connected to a 2.5G Ethernet jet on an Asus router

### Description of your environment

Not Applicable since the link is not up

### Output of ``dmesg`` command
<details>
<summary>dmesg</summary>

```shell
$ dmesg
...
[   94.870627] ext2: synoboot2 mounted, process=synopkicompatsy
[   94.883089] synoboot2 unmounted, process=synopkicompatsy
[   96.380825] iSCSI:target_core_rodsp_server.c:1025:rodsp_server_init RODSP server started, login_key(8d8f259a534e).
[   96.410611] iSCSI:extent_pool.c:766:ep_init syno_extent_pool successfully initialized
[   96.444791] iSCSI:target_core_device.c:612:se_dev_align_max_sectors Rounding down aligned max_sectors from 4294967295 to 4294967288
[   96.458042] iSCSI:target_core_configfs.c:5763:target_init_dbroot db_root: cannot open: /etc/target
[   96.468472] iSCSI:target_core_lunbackup.c:366:init_io_buffer_head 2048 buffers allocated, total 8388608 bytes successfully
[   96.551910] iSCSI:target_core_file.c:152:fd_attach_hba RODSP plugin for fileio is enabled.
[   96.561179] iSCSI:target_core_file.c:159:fd_attach_hba ODX Token Manager is enabled.
[   96.569872] iSCSI:target_core_multi_file.c:91:fd_attach_hba RODSP plugin for multifile is enabled.
[   96.579902] iSCSI:target_core_ep.c:795:ep_attach_hba RODSP plugin for epio is enabled.
[   96.588777] iSCSI:target_core_ep.c:802:ep_attach_hba ODX Token Manager is enabled.
[   96.757876] workqueue: max_active 1024 requested for vhost_scsi is out of range, clamping between 1 and 512
[   97.891297] ext2: synoboot1 mounted, process=syno_bios_log_c
[   97.934370] vfat: synoboot1 mounted, process=syno_bios_log_c
[   98.007505] synoboot1 unmounted, process=syno_bios_log_c
[   98.876121] findhostd uses obsolete (PF_INET,SOCK_PACKET)
[   99.589126] mii: module verification failed: signature and/or required key missing - tainting kernel
[   99.692652] usbcore: registered new interface driver aqc111
[   99.705774] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:80:8e:e4
[  100.000277] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  100.086037] device ovs_eth2 entered promiscuous mode
[  100.091614] audit_printk_skb: 84 callbacks suppressed
[  100.097281] audit: type=1700 audit(1695311632.940:82): dev=ovs_eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  100.120524] device eth2 entered promiscuous mode
[  100.125725] audit: type=1700 audit(1695311632.974:83): dev=eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  100.141372] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[  100.945014] audit: type=1325 audit(1695311633.793:84): table=filter family=2 entries=16
[  100.958809] audit: type=1325 audit(1695311633.807:85): table=filter family=2 entries=17
[  101.852814] audit: type=1325 audit(1695311634.701:86): table=filter family=2 entries=18
[  101.877319] audit: type=1325 audit(1695311634.725:87): table=filter family=2 entries=17
[  102.571688] xhci_hcd 0000:00:15.0: device is plugged out, empty URBs
[... 187 lines cut ...]
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:80:8E:E4
          inet addr:192.168.50.157  Bcast:192.168.50.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
</details>

---

@bb-qq Thanks for your good work! If you'd have time looking into this issue that be great!

udmada, 2023-09-21:
### Content of ``/var/packages/aqc111/var/log/start-stop-status.log``
<details>
<summary>cat /var/packages/aqc111/var/log/start-stop-status.log</summary>

```shell
$ cat /var/packages/aqc111/var/log/start-stop-status.log

+ date
Fri Sep 22 07:10:12 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Fri Sep 22 07:10:38 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Fri Sep 22 08:07:00 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Fri Sep 22 08:07:28 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
[... 937 lines cut ...]
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
+ echo ----------
----------
+ date
Fri Sep 22 08:26:59 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo status
status
+ case $1 in
+ /sbin/lsmod
+ grep aqc111
aqc111                 18841  0
usbnet                 20817  1 aqc111
usbcore               209065  12 uas,etxhci_hcd,usblp,uhci_hcd,usb_storage,aqc111,ehci_hcd,ehci_pci,usbhid,usbnet,xhci_hcd,xhci_pci
+ exit 0
```
</details>

udmada, 2023-09-21:
### Content of ``/var/log/packages/aqc111.log``
<details>
<summary>cat /var/log/packages/aqc111.log</summary>

```shell
$ cat /var/log/packages/aqc111.log
2023/09/22 03:53:52	(system) trigger aqc111 1.3.3.0-12 Begin start-stop-status start
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/volume1/@appstore/aqc111/aqc111
+ log_root=/var/packages/aqc111/var/log
+ mkdir -p /var/packages/aqc111/var/log
+ exec
++ tee -a /var/packages/aqc111/var/log/start-stop-status.log
+ echo ----------
----------
+ date
Fri Sep 22 03:53:52 NZST 2023
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/aqc111/scripts/apply-memory-setting ']'
+ sh /var/packages/aqc111/scripts/apply-memory-setting
vm.min_free_kbytes = 524288
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
+ echo -1
++ count_usb_interfaces
++ wc -l
++ ls -l /sys/class/net/
++ grep /usb
+ initial_count=0
+ '[' -e /volume1/@appstore/aqc111/aqc111/mii.ko ']'
+ /sbin/insmod /volume1/@appstore/aqc111/aqc111/mii.ko
+ '[' -e /volume1/@appstore/aqc111/aqc111/usbnet.ko ']'
+ /sbin/insmod /volume1/@appstore/aqc111/aqc111/usbnet.ko
+ /sbin/insmod /volume1/@appstore/aqc111/aqc111/aqc111.ko
+ '[' -r /usr/lib/udev/rules.d/51-usb-aqc111-net.rules ']'
++ seq 1 60
+ for i in `seq 1 60`
++ count_usb_interfaces
++ grep /usb
++ wc -l
++ ls -l /sys/class/net/
+ '[' 1 -ne 0 ']'
+ break
+ set_interfaces up
+ action=up
+ target_interface_name=
++ ls /sys/class/net
+ all_interface_names='eth0
eth1
eth2
lo
ovs-system
ovs_eth0
ovs_eth1
sit0
syno_ovs_bonds'
+ '[' -n ']'
+ '[' '!' -e /sys/class/net/ ']'
+ for interface_name in ${all_interface_names}
+ [[ -n '' ]]
+ '[' '!' -e /sys/class/net/eth0/device/driver ']'
[... 159 lines cut ...]
    inet 192.168.50.157/24 brd 192.168.50.255 scope global ovs_eth2
       valid_lft forever preferred_lft forever
11: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 02:42:0b:2a:c5:e4 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
12: eth2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 24:5e:be:80:8e:e4 brd ff:ff:ff:ff:ff:ff
```
</details>

### Output of ``ip route`` command
<details>
<summary>ip route</summary>

```shell
$ ip route

default via 192.168.50.1 dev ovs_eth0  src 192.168.50.154
169.254.0.0/16 dev ovs_eth1  proto kernel  scope link  src 169.254.15.203 dead linkdown
172.17.0.0/16 dev docker0  proto kernel  scope link  src 172.17.0.1 linkdown
192.168.50.0/24 dev ovs_eth0  proto kernel  scope link  src 192.168.50.154
192.168.50.0/24 dev ovs_eth2  proto kernel  scope link  src 192.168.50.157 linkdown
```
</details>

bb-qq, 2023-10-09:
There are many reports of working with the DS920+, so there may be a problem in the environment or on the QNA-UC5G1T side.

Does it work fine when QNA-UC5G1T is connected to a PC using the same cable?