bb-qq/aqc111 #123: Active/Passive Bond does not work

marcel151, 2023-10-10:
### Description of the problem

I am following up this issue:
https://github.com/bb-qq/aqc111/issues/17

Since I can't reopen that I'll open this issue.

I can't use active/passive failsafe with one of the internal NICs. The bond is created, but not used correctly. I just have Gigabit speeds. When I disconnect the Gigabit connection while using failsave bond, it does not connect at all. 5 Gbit/s works fine when using the interface on it's own.

### Description of your products

> - Synology DS920+
>     - Linux Synology 4.4.302+ #64570 SMP Thu Jul 20 00:06:38 CST 2023 x86_64 GNU/Linux synology_geminilake_920+
>     - DSM version 7.2-64570 Update 3
> - QNAP QNA-UC5G1T

### Description of your environment

> - Connected to a Ubiquiti UniFi Dream Machine SE using SFP+ to Ethernet
       - SFP+: UACC-CM-RJ45-MG
> - 1.5m Cat. 7

### Output of ``dmesg`` command

````
Since I had too many characters, I'm using justpaste.it for dmesg, hope that's okay.
(https://justpaste.it/324rg)
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__1-2         05e3:0610:4143 09  2.10  480MBit/s 100mA 1IF  (GenesysLogic USB2.0 Hub ffffffd6ffffffa3ffffffebffffffcb) hub
    |__1-2.2     051d:0002:0090 00  2.00   12MBit/s 2mA 1IF  (American Power Conversion Back-UPS ES 850G2 FW:938.a2 .I U 5B2111T00004  )
  |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F0005DAB913F223)
|__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub
  |__2-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP QNAP QNA-UC5G1T USB to 5GbE Adapter 34I04709)
  |__2-2         05e3:0617:4143 09  3.00 5000MBit/s 0mA 1IF  (GenesysLogic USB3.0 Hub ffffffd6ffffffa3ffffffebffffffc0) hub
````

### Output of ``ifconfig -a`` command

````
docker0   Link encap:Ethernet  HWaddr 02:42:E1:23:87:BD  
          inet addr:172.17.0.1  Bcast:172.17.255.255  Mask:255.255.0.0
          inet6 addr: fe80::42:e1ff:fe23:87bd/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13882 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10402 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:1228602 (1.1 MiB)  TX bytes:2169486 (2.0 MiB)

docker14e Link encap:Ethernet  HWaddr DE:37:3F:2A:BD:9A  
          inet6 addr: fe80::dc37:3fff:fe2a:bd9a/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:11 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:918 (918.0 B)

dockerbab Link encap:Ethernet  HWaddr 52:EA:00:8C:78:D9  
          inet6 addr: fe80::50ea:ff:fe8c:78d9/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:13882 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10402 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:1422950 (1.3 MiB)  TX bytes:2169486 (2.0 MiB)

eth0      Link encap:Ethernet  HWaddr 00:11:32:CD:DC:F5  
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:6777244 errors:0 dropped:0 overruns:0 frame:0
          TX packets:7232227 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6856827303 (6.3 GiB)  TX bytes:7586961777 (7.0 GiB)
          Interrupt:97 base 0x6000  

eth1      Link encap:Ethernet  HWaddr 00:11:32:CD:DC:F6  
          UP BROADCAST SLAVE MULTICAST  MTU:9000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Interrupt:96 base 0xe000  

eth2      Link encap:Ethernet  HWaddr 24:5E:BE:81:03:EF  
          inet6 addr: fe80::265e:beff:fe81:3ef/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:962 errors:1 dropped:0 overruns:0 frame:0
          TX packets:16 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:196290 (191.6 KiB)  TX bytes:5064 (4.9 KiB)

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:579954 errors:0 dropped:0 overruns:0 frame:0
          TX packets:579954 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:220306811 (210.1 MiB)  TX bytes:220306811 (210.1 MiB)

ovs-syste Link encap:Ethernet  HWaddr 7E:57:81:85:F8:06  
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_bond0 Link encap:Ethernet  HWaddr 00:11:32:CD:DC:F5  
          inet addr:192.168.2.10  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2015388 errors:0 dropped:0 overruns:0 frame:0
          TX packets:801321 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:2482970033 (2.3 GiB)  TX bytes:2214751983 (2.0 GiB)

ovs_eth1  Link encap:Ethernet  HWaddr 00:11:32:CD:DC:F6  
          inet addr:169.254.47.155  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:9000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
````

BiffaZ, 2023-12-31:
Have you tried the solution here?: https://community.synology.com/enu/forum/1/post/129091

bb-qq, 2024-06-15:
This is a limitation of the current driver.

Under the current mechanism, drivers are loaded after the package system is ready, so LACP BOND, which is prepared at an earlier stage, is not restored at startup.

If the drivers are loaded at the same time as other kernel modules are loaded at startup, the LACP configuration can be enabled. But I have not found a good way to do that.

related: https://github.com/bb-qq/r8152/issues/381