bb-qq/aqc111 #124: AQC111 driver with QNAP QNA-UC5G1T USB 3.0 to 5Gbe Adapter always stops working after some time

salsafunatiker, 2023-10-16:
I have Synology DS720+.

Installation works fine for QNAP QNA-UC5G1T and with iperf3 I get throughput like 3.2 GBit.
After transferring lots of data, it then stops working. I need to stop AQC111 driver and start again.

In contrast to QNAP 5 Gbit adapter, the 2.5 Gbit adapter for R8152 always works fine and stable.

Who can help with my 5 Gbit issue? Does anyone experience the same?

mattrpav, 2023-10-23:
Most likely, it is shutting down due to heat. This aqc111 driver supports setting priv flags to use less power, and disable the thermal throttling and works for the most part.

```
    ethtool --set-priv-flags eth2 "Thermal throttling" off
    ethtool --set-priv-flags eth2 "Low Power 5G" on
```

@bb-qq it'd be great if these features were merged into the mainline kernel. Seems like it'd make sense to have those without having to patch the system after every kernel update. Thoughts?

salsafunatiker, 2023-10-24:
Thank you for the answer!
Here is described how to do it:
https://github.com/bb-qq/aqc111/issues/7

With this, the QNAP 5 Gbit adapter seems to work stable.

However, some general words, the 5 Gbit over USB has limitations:
- it gets really hot
- the data rate is not 5 Gbps!
- Instead it does not reach more than 3.5 Gbps
- Buying 2.5 Gbps gives you max 2.35 Gbps
- Buying 5 Gbps gives you max 3.5 Gbps - in ideal condition

closed 2023-10-26 by salsafunatiker

bb-qq, 2023-10-29:
> it'd be great if these features were merged into the mainline kernel. Seems like it'd make sense to have those without having to patch the system after every kernel update. Thoughts?

The source code my driver is based on is provided by the vendor (Aquantia/Marvell), and the Linux kernel driver is committed by the vendor as well.
The code on the Linux kernel side seems to be still being updated, so there may be a reason why these flag settings are not supported.

mattrpav, 2023-10-29:
@bb-qq thank you for that info. Where is the appropriate place to request the feature? Here or Linux kernel issue tracker?