bb-qq/aqc111 #125: QNAP TS-233

pihokkerr, 2023-10-30:
### Unsupported adapter 

Would it be possible to implement the project for unsupported QNAP NAS as well? When the TS-233 NAS was released, the QNA-UC5G1T product was still among the compatible devices, but a few months later it was removed, and it hasn't worked since.

### Description of your products

> QNAP TS-233
> CPU: 64-bit ARM Cortex-A55
> QTS 5.1.2
> QNA-UC5G1T

bb-qq, 2023-11-03:
I do not have the skills to develop a driver for QNAP. And if they dare to stop supporting the device, I guess there is a technical reason.

Dmedia3412, 2023-11-28:
Add in autorun.sh

modprobe aqc111
ethtool --set-priv-flags eth2 "Thermal throttling" off
ethtool --set-priv-flags eth2 "Low Power 5G" on

bb-qq, 2024-01-06:
I am sorry, but there is nothing more I can do for you, so I am closing this issue.

closed 2024-01-06 by bb-qq