bb-qq/aqc111 #128: QNA-UC5G1T performance 2.4 Gbps >> 1.4 Gbps after QNAP firmware update 3.1.6

FredDL31, 2024-01-24:
### Description of the problem

Transfer speed was ~2.4 Gbits/sec, then after updating QNA-UC5G1T to last firmware (3.1.6) expecting improving speed, finally speed fall down to 1.4 Gbits/sec ... No other change in HW/SW configuration.

iperf3 before firmware update

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   701 MBytes   588 Mbits/sec                  sender
[  4]   0.00-10.00  sec   701 MBytes   588 Mbits/sec                  receiver
[  6]   0.00-10.00  sec   739 MBytes   620 Mbits/sec                  sender
[  6]   0.00-10.00  sec   739 MBytes   620 Mbits/sec                  receiver
[  8]   0.00-10.00  sec   726 MBytes   609 Mbits/sec                  sender
[  8]   0.00-10.00  sec   726 MBytes   609 Mbits/sec                  receiver
[ 10]   0.00-10.00  sec   743 MBytes   623 Mbits/sec                  sender
[ 10]   0.00-10.00  sec   743 MBytes   623 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  2.84 GBytes  2.44 Gbits/sec                  sender
[SUM]   0.00-10.00  sec  2.84 GBytes  2.44 Gbits/sec                  receiver

iperf3 after firmware update

[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec   282 MBytes   237 Mbits/sec                  sender
[  4]   0.00-10.00  sec   282 MBytes   237 Mbits/sec                  receiver
[  6]   0.00-10.00  sec   502 MBytes   421 Mbits/sec                  sender
[  6]   0.00-10.00  sec   502 MBytes   421 Mbits/sec                  receiver
[  8]   0.00-10.00  sec   573 MBytes   481 Mbits/sec                  sender
[  8]   0.00-10.00  sec   573 MBytes   481 Mbits/sec                  receiver
[ 10]   0.00-10.00  sec   285 MBytes   239 Mbits/sec                  sender
[ 10]   0.00-10.00  sec   285 MBytes   239 Mbits/sec                  receiver
[SUM]   0.00-10.00  sec  1.60 GBytes  1.38 Gbits/sec                  sender
[SUM]   0.00-10.00  sec  1.60 GBytes  1.38 Gbits/sec                  receiver

Rq: acq111 driver for DSM 7.2 broadwellnk not provided on GitHub, compiled from sources

### Description of your products

>     - Xpenology DS3622XS on baremetal HP Micro Gen8 w/QNAP QNA-UC5G1T
>     - Linux 4.4.302+ #69057 SMP Fri Jan 12 17:02:26 CST 2024 x86_64 GNU/Linux synology_broadwellnk_3622xs+
>     - DSM 7.2.1-69057 Update 4
>     - QNAP QNA-UC5G1T

### Description of your environment

>     - connect using a hub
>     - PC: Intel X520-DA1
>     - Hub: CISCO/Linksys LGS328C
>     - cable category Cat.6

### Output of ``dmesg`` command

````
[  359.153170] aqc111 5-1:1.0 eth2: register 'aqc111' at usb-0000:04:00.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:4d:a7:0b
[  359.153481] usbcore: registered new interface driver aqc111
[  359.422395] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[  359.529220] device ovs_eth2 entered promiscuous mode
[  359.529350] audit_printk_skb: 609 callbacks suppressed
[  359.529469] audit: type=1700 audit(1706093376.846:3183): dev=ovs_eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  359.535920] device eth2 entered promiscuous mode
[  359.536034] audit: type=1700 audit(1706093376.853:3184): dev=eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[  359.538020] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[  366.877831] aqc111 5-1:1.0 eth2: Link Speed 5000, USB 3
[  366.884645] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready
````

### Output of ``lsusb`` command

````
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1a.0) hub
  |__1-1         8087:0024:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffad) hub
    |__1-1.5     0463:ffff:0100 00  1.10  1.5MBit/s 20mA 1IF  (EATON Ellipse PRO P360G45BA6)
|__usb2          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (ehci_hcd 0000:00:1d.0) hub
  |__2-1         8087:0024:0000 09  2.00  480MBit/s 0mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa0) hub
    |__2-1.3     0424:2660:0801 09  2.00  480MBit/s 2mA 1IF  ( ffffffd1ffffffb2ffffffdbffffffa1) hub
      |__2-1.3.1 f400:f400:0210 00  2.00  480MBit/s 96mA 1IF  (Generic Ultra Fast Media Reader 000002660A01)
|__usb3          1d6b:0001:0404 09  1.10   12MBit/s 0mA 1IF  (uhci_hcd 0000:01:00.4) hub
|__usb4          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:04:00.0) hub
|__usb5          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:04:00.0) hub
  |__5-1         1c04:0015:0101 00  3.20 5000MBit/s 896mA 1IF  (QNAP▒ Pacific▒ 02I08969)
````

### Output of ``ifconfig -a`` command

````
ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:4D:A7:0B
          inet addr:192.168.1.85  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe4d:a70b/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:9000  Metric:1
          RX packets:1877154 errors:0 dropped:0 overruns:0 frame:0
          TX packets:540491 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:2420592616 (2.2 GiB)  TX bytes:10727146890 (9.9 GiB)
````

bb-qq, 2024-02-10:
What will the speed be if I use the same configuration of Hub and cables but change only the QNA-UC5G1T to connect to a Windows PC?

FredDL31, 2024-02-10:
This is good idea, and with Windows PC I performed tests, and speed is correct:

D:\System\iperf-3.1.3-win64>iperf3 -c 192.168.1.85
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-1.00   sec   281 MBytes  2.36 Gbits/sec
[  4]   1.00-2.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   2.00-3.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   3.00-4.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   4.00-5.00   sec   280 MBytes  2.34 Gbits/sec
[  4]   5.00-6.00   sec   279 MBytes  2.34 Gbits/sec
[  4]   6.00-7.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   7.00-8.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   8.00-9.00   sec   280 MBytes  2.35 Gbits/sec
[  4]   9.00-10.00  sec   280 MBytes  2.35 Gbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth
[  4]   0.00-10.00  sec  2.74 GBytes  2.35 Gbits/sec                  sender
[  4]   0.00-10.00  sec  2.74 GBytes  2.35 Gbits/sec                  receiver

Btw, it is something wrt last firmware (but from 2020 !) vs. driver ?

Thanks.

bb-qq, 2024-02-26:
I have already updated the firmware in my environment and the performance is fine. I also have low power mode enabled.

When you measure the speed on a Windows PC, are all other conditions the same, such as MTU, opposing devices, switches along the way, etc.?

FredDL31, 2024-02-26:
Sure, strictly same configuration ... tested on 3 differents PC, all on same switch as the NAS, with standard MTU to avoid pb.

The 3 PC shown ~2.4 Gbits/sec, where the NAS now (after firmware update) show 1.6 Gbits/sec.

Btw, it was working fine (except a "poor" 2.44 Gbits/sec expected better), and then only firmware was updated in the whole HW/SW configuration, with consequence performed bad as ever on NAS ... but stable on PC.

bb-qq, 2024-06-15:
Since then I have tried to find out where the stock firmware is, but unfortunately I could not find it.

This adapter is inherently capable of more than 4Gbps, we may as well forget about Synology and investigate what the bottleneck is.
For example, change the switch to a direct connection, adjust the MTU values of the devices related, etc.