bb-qq/aqc111 #129: QNA-UC5G1T becoming unreliable after updating to DSM 7.2.1-69057 Update 5 (from Update 4)

marcel151, 2024-04-29:
### Description of the problem

Connection becomes unreliable, it works at first and after a few days my DS920+ is not reachable anymore with 5Gb IP. I'll have to stop and start the driver again. After that it works again and suddenly stops working. Totally unreliable.

### Description of your products

> Synology DS920+
>     Linux Synology 4.4.302+ #69057 SMP Fri Jan 12 17:02:28 CST 2024 x86_64 GNU/Linux synology_geminilake_920+
>     DSM version 7.2.1-69057 Update 5
>     QNAP QNA-UC5G1T

### Description of your environment

> Connected to my UDM SE via switch SODOLA SL-SWTG124AS-D
>     Synology -> QNA-UC5G1T -> UniFi UACC-CM-RJ45-MG -> SODOLA SL-SWTG124AS-D -> UniFi UACC-DAC-SFP10-0.5M -> UniFi Dream Machine SE 
>     Cat.7 Cable

### Output of ``dmesg`` command

````
[379658.532700] audit: type=1700 audit(1714402835.810:121): dev=eth2 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[379658.550473] Module [aqc111] is removed.
[379658.554996] usbcore: deregistering interface driver aqc111
[379658.561357] aqc111 2-1:1.0 eth2: unregister 'aqc111' usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[379658.590846] Module [usbnet] is removed.
[379658.606192] Module [mii] is removed.
[379658.810226] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.2.11:3260 on iSCSI/TCP
[379658.822819] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: [fe80::265e:beff:fe81:3ef]:3260 on iSCSI/TCP
[379664.007396] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:81:03:ef
[379664.020519] usbcore: registered new interface driver aqc111
[379664.049870] Synotify use 16384 event queue size
[379664.284865] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[379664.321991] device ovs_eth2 entered promiscuous mode
[379664.327674] audit: type=1700 audit(1714402841.606:122): dev=ovs_eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[379664.354942] device eth2 entered promiscuous mode
[379664.360244] audit: type=1700 audit(1714402841.638:123): dev=eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[379664.376431] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[379665.681665] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: 192.168.2.11:3260 on iSCSI/TCP
[379668.131318] Synotify use 16384 event queue size
[379672.079063] aqc111 2-1:1.0 eth2: Link Speed 5000, USB 3
[379672.094503] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready
[379672.314766] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: [fe80::265e:beff:fe81:3ef]:3260 on iSCSI/TCP
[379674.516475] Synotify use 16384 event queue size
[379679.869111] Synotify use 16384 event queue size
[380526.818953] device ovs_eth2 left promiscuous mode
[380526.824410] audit: type=1700 audit(1714403704.272:124): dev=ovs_eth2 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[380526.844334] device eth2 left promiscuous mode
[380526.849355] audit: type=1700 audit(1714403704.297:125): dev=eth2 prom=0 old_prom=256 auid=4294967295 uid=0 gid=0 ses=4294967295
[380526.866140] Module [aqc111] is removed.
[380526.870629] usbcore: deregistering interface driver aqc111
[380526.876920] aqc111 2-1:1.0 eth2: unregister 'aqc111' usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter
[380526.908699] Module [usbnet] is removed.
[380526.918640] Module [mii] is removed.
[380527.033306] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: 192.168.2.11:3260 on iSCSI/TCP
[380527.046367] iSCSI:iscsi_target.c:642:iscsit_del_np CORE[0] - Removed Network Portal: [fe80::265e:beff:fe81:3ef]:3260 on iSCSI/TCP
[380530.228554] aqc111 2-1:1.0 eth2: register 'aqc111' at usb-0000:00:15.0-1, QNAP QNA-UC5G1T USB to 5GbE Adapter, 24:5e:be:81:03:ef
[380530.241677] usbcore: registered new interface driver aqc111
[380530.497616] IPv6: ADDRCONF(NETDEV_UP): eth2: link is not ready
[380530.580360] device ovs_eth2 entered promiscuous mode
[380530.586032] audit: type=1700 audit(1714403708.034:126): dev=ovs_eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[380530.614196] device eth2 entered promiscuous mode
[380530.619475] audit: type=1700 audit(1714403708.067:127): dev=eth2 prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[380530.638655] IPv6: ADDRCONF(NETDEV_UP): ovs_eth2: link is not ready
[380531.605977] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: 192.168.2.11:3260 on iSCSI/TCP
[380532.259056] Synotify use 16384 event queue size
[380535.003201] Synotify use 16384 event queue size
[380539.119056] aqc111 2-1:1.0 eth2: Link Speed 5000, USB 3
[380539.134571] IPv6: ADDRCONF(NETDEV_CHANGE): ovs_eth2: link becomes ready
[380539.273238] iSCSI:iscsi_target.c:544:iscsit_add_np CORE[0] - Added Network Portal: [fe80::265e:beff:fe81:3ef]:3260 on iSCSI/TCP
[380541.482427] Synotify use 16384 event queue size
[... 143 lines cut ...]
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

ovs_eth2  Link encap:Ethernet  HWaddr 24:5E:BE:81:03:EF
          inet addr:192.168.2.11  Bcast:192.168.2.255  Mask:255.255.255.0
          inet6 addr: fe80::265e:beff:fe81:3ef/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:65250 errors:0 dropped:0 overruns:0 frame:0
          TX packets:61537 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:10854180 (10.3 MiB)  TX bytes:1640684521 (1.5 GiB)

sit0      Link encap:IPv6-in-IPv4
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

tap021132 Link encap:Ethernet  HWaddr 6E:90:A6:8D:FE:AA
          UP BROADCAST RUNNING SLAVE MULTICAST  MTU:1500  Metric:1
          RX packets:8143925 errors:0 dropped:0 overruns:0 frame:0
          TX packets:9880888 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:20911787662 (19.4 GiB)  TX bytes:1500788940 (1.3 GiB)
````

bb-qq, 2024-06-15:
I don't see any reports of instability from this DSM version, and since there is no difference in my environment, I suspect that something else may be affecting it.

Could you please look into the problem using this page?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#stability-issues

bb-qq, 2025-11-16:
Also, please try disconnecting other USB devices and see how it goes.