bb-qq/aqc111 #136: Can't install driver on DS920+ DSM 7.2.1-69057 Update 5

jameslashmar, 2024-09-06:
Description of the problem
Driver was working as it was installed prior to DSM update. It has actually been super solid over the past year!

Removing and reinstalling now is broke in a loop. Driver fails to install, after running sudo command, the driver is left in an error state and cannot be install over the top of itself. The error states, "failed to repair aqc111 driver'. The only option is to remove the spk.

Have tried system reboot, the troubleshooting guide and stopping all docker containers, uninstalling Virtual Machine Manager & Container manager.

Using QNA-UC5G1T with FW 3.1.6

Output of dmesg command
attached

[dmesg.txt](https://github.com/user-attachments/files/16906973/dmesg.txt)

Output of lsusb command
|__usb1          1d6b:0002:0404 09  2.00  480MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub |__1-2         0764:0501:0001 00  1.10  1.5MBit/s 50mA 1IF  (CPS BR700ELCD fffffff0ffffffb3ffffffffffffffba) |__1-4         f400:f400:0100 00  2.00  480MBit/s 200mA 1IF  (Synology DiskStation 7F000ACB3B13F274) |__usb2          1d6b:0003:0404 09  3.00 5000MBit/s 0mA 1IF  (Linux 4.4.302+ xhci-hcd xHCI Host Controller 0000:00:15.0) hub

Output of ifconfig -a command
`docker0 Link encap:Ethernet HWaddr 02:42:77:EC:7E:AB
inet addr:172.17.0.1 Bcast:172.17.255.255 Mask:255.255.0.0
inet6 addr: fe80::42:77ff:feec:7eab/64 Scope:Link
UP BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:909 errors:0 dropped:0 overruns:0 frame:0
TX packets:1016 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:1019290 (995.4 KiB) TX bytes:463716 (452.8 KiB)

docker-84 Link encap:Ethernet HWaddr 02:42:6D:35:AB:E8
inet addr:192.168.48.1 Bcast:192.168.63.255 Mask:255.255.240.0
inet6 addr: fe80::42:6dff:fe35:abe8/64 Scope:Link
UP BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:5 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:0 (0.0 B) TX bytes:438 (438.0 B)

eth0 Link encap:Ethernet HWaddr 00:11:32:DE:96:B1
UP BROADCAST SLAVE MULTICAST MTU:9000 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)
Interrupt:97 base 0x2000

eth1 Link encap:Ethernet HWaddr 00:11:32:DE:96:B2
UP BROADCAST RUNNING SLAVE MULTICAST MTU:9000 Metric:1
RX packets:143253 errors:0 dropped:0 overruns:0 frame:0
TX packets:116528 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000
RX bytes:58882901 (56.1 MiB) TX bytes:40144341 (38.2 MiB)
Interrupt:96

lo Link encap:Local Loopback
inet addr:127.0.0.1 Mask:255.0.0.0
inet6 addr: ::1/128 Scope:Host
UP LOOPBACK RUNNING MTU:65536 Metric:1
RX packets:35871 errors:0 dropped:0 overruns:0 frame:0
TX packets:35871 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1
RX bytes:8702730 (8.2 MiB) TX bytes:8702730 (8.2 MiB)

ovs-syste Link encap:Ethernet HWaddr 4E:BD:8A:66:7E:2E
BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

ovs_eth0 Link encap:Ethernet HWaddr 00:11:32:DE:96:B1
inet addr:169.254.80.15 Bcast:169.254.255.255 Mask:255.255.0.0
UP BROADCAST MULTICAST MTU:9000 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

ovs_eth1 Link encap:Ethernet HWaddr 00:11:32:DE:96:B2
inet addr:10.198.6.150 Bcast:10.198.6.255 Mask:255.255.255.0
UP BROADCAST RUNNING MULTICAST MTU:1500 Metric:1
RX packets:128915 errors:0 dropped:0 overruns:0 frame:0
TX packets:88758 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1
RX bytes:53937070 (51.4 MiB) TX bytes:39244433 (37.4 MiB)

sit0 Link encap:IPv6-in-IPv4
NOARP MTU:1480 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)`

jameslashmar, 2024-09-06:
Tried to install directly via SSH using synopkg and received

```
ash-4.4#  synopkg install aqc111-geminilake-1.3.3.0-12_7.2.spk
{"error":{"code":0},"results":[{"action":"repair","beta":false,"betaIncoming":false,"broken_by":"install_corruption","error":{"code":278,"description":"\"failed to run postinst script\""},"finished":true,"installReboot":false,"installing":true,"language":"enu","last_stage":"postinstall","package":"aqc111","packageName":"AQC111 driver","pid":2260,"scripts":[{"code":0,"message":"","type":"preupgrade"},{"code":0,"message":"","type":"preuninst"},{"code":0,"message":"","type":"postuninst"},{"code":0,"message":"","type":"preinst"},{"code":5,"message":"","type":"postinst"}],"spk":"aqc111-geminilake-1.3.3.0-12_7.2.spk","stage":"install_failed_and_stopped","status":"brokenash-4.4#
```

bb-qq, 2024-09-19:
Could you attach the entire log of /volume1/@appdata/aqc111/log/start-stop-status.log?
I need a `start` entry, but the above only includes a `stop` entry.

jameslashmar, 2024-09-24:
Apologies for the delay, i've been on holiday - here you go!

[start-stop.log](https://github.com/user-attachments/files/17113442/start-stop.log)
[synopkg.log](https://github.com/user-attachments/files/17113443/synopkg.log)

bb-qq, 2024-09-24:
```
+ synonet --dhcp ovs_eth2
synonet.c:393 Failed to read '/sys/class/net/ovs_eth2/speed'
Host Name: Oracle
Network interface: ovs_eth2
DHCP
IP:
Mask:
Gateway: 10.198.6.1
DNS: 208.67.222.222
MTU Setting: 9000
```
This log indicates that something is preventing DHCP from working properly.

Please follow this documentation and enable hot-plugging to start the driver. Then, could you try to assign a fixed address after that?
https://github.com/bb-qq/r8152/wiki/Troubleshooting#hot-plugging-does-not-work

jameslashmar, 2024-09-24:
OK that did something. The lights on the device are now lit up and the adaptor is showing under network.  I've manually set it's ip etc.

Weirdly, despite the .spk being stuck, the adaptor is still listed.

[image]

Here's the ssh log from me starting with the .spk completely uninstalled.

```
jay@Oracle:/var/packages/aqc111/target/aqc111$ sudo bash /var/packages/aqc111/scripts/install-udev-rules
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
bash: /var/packages/aqc111/scripts/install-udev-rules: No such file or directory
jay@Oracle:/var/packages/aqc111/target/aqc111$ sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su
install: cannot stat '/var/packages/aqc111/target/aqc111/spk_su': No such file or directory
jay@Oracle:/var/packages/aqc111/target/aqc111$ sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su
jay@Oracle:/var/packages/aqc111/target/aqc111$ ls
jay@Oracle:/var/packages/aqc111/target/aqc111$ sudo bash /var/packages/aqc111/scripts/install-udev-rules
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
ln: failed to create symbolic link '/usr/lib/udev/rules.d/51-usb-aqc111-net.rules': File exists
jay@Oracle:/var/packages/aqc111/target/aqc111$ sudo /var/packages/aqc111/scripts/start-stop-status start
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
++ dirname /var/packages/aqc111/scripts/start-stop-status
+ script_root=/var/packages/aqc111/scripts
++ realpath -m /var/packages/aqc111/scripts/..
+ package_root=/var/packages/aqc111
++ basename /var/packages/aqc111
+ driver_name=aqc111
+ driver_root=/var/packages/aqc111/target/aqc111
+ log_root=/var/packages/aqc111/var/log
+ mkdir -p /var/packages/aqc111/var/log
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
+ exec
++ tee -a /var/packages/aqc111/var/log/start-stop-status.log
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
+ echo ----------
----------
+ date
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
Tue Sep 24 01:58:48 PM IST 2024
+ grep -E '^(version|create_time)' /var/packages/aqc111/INFO
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
version="1.3.3.0-12"
create_time="20230715-10:16:47"
+ echo start
start
+ case $1 in
+ '[' -e /var/packages/aqc111/scripts/apply-memory-setting ']'
+ sh /var/packages/aqc111/scripts/apply-memory-setting
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
shell-init: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
+ '[' -w /sys/module/usbcore/parameters/autosuspend ']'
+ echo -1
++ count_usb_interfaces
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
++ wc -l
++ grep /usb
++ ls -l /sys/class/net/
+ initial_count=1
+ '[' -e /var/packages/aqc111/target/aqc111/mii.ko ']'
+ /sbin/insmod /var/packages/aqc111/target/aqc111/mii.ko
job-working-directory: error retrieving current directory: getcwd: cannot access parent directories: No such file or directory
insmod: ERROR: could not insert module /var/packages/aqc111/target/aqc111/mii.ko: File exists
jay@Oracle:/var/packages/aqc111/target/aqc111$
```

bb-qq, 2024-10-05:
Is the situation the same after rebooting?
I have vSwitch and those apps enabled in my environment, so that probably has nothing to do with it.

jameslashmar, 2024-10-15:
Yeah, i had tried that.

I just tried again too. This time i noticed the system checkup was flagging a few things in terms of security like scripts etc to boot with and also entware.  I fixed all off these, removed the .spk and rebooted.  After the reboot this time the adaptor disappeared.

I tried to install the spk again and it failed.  I closed out of the install settings (otherwise you are stuck in the repair loop), then uninstalled the spk, rand the sudo command and tried manual install again. It failed. So i tried running the sudo command while in the install dialogue after it fails, but then it still fails again on the second try.

If i run the sudo command with -v, this is what i get

```
jay@Oracle:/$ sudo install -m 4755 -o root -D /var/packages/aqc111/target/aqc111/spk_su /opt/sbin/spk_su -v
removed '/opt/sbin/spk_su/spk_su'
'/var/packages/aqc111/target/aqc111/spk_su' -> '/opt/sbin/spk_su/spk_su'
```

closed 2024-10-23 by jameslashmar