bb-qq/aqc111 #137: DS918+ and driver aqc111-apollolake-1.3.3.0-12_7.2.spk on DSM7.2.2-72806 only 1000 MBit/s

Merlin-2504, 2024-10-31:
I tested the driver aqc111-apollolake-1.3.3.0-12_7.2.spk and the driver r8152-apollolake-2.18.1-2_7.2.spk on my DS918 with DSM7.2.2-72806.
I only get 1000 Mbit/s? What am I doing wrong?
The adapter is the WAVLINK USB C to 5G Ethernet adapter, USB C adapter with 5000 Mbit/s

Displayel, 2024-10-31:
And you think, with "What am I doing wrong?" You magically get the reply you need?

Merlin-2504, 2024-10-31:
[image]
[image]
[image]

bb-qq, 2024-11-20:
Please check this article:
https://github.com/bb-qq/r8152/wiki/Troubleshooting#check-the-protocol-version-of-the-usb-connection